zephyrproject-rtos/zephyr #21148: nrf51: uart_1 does not compile

aep, 2019-12-03:
nrf52840_pca10056 doesn't seem to compile with uart1 enabled

i suspect this dts is wrong in boards/arm/nrf52840_pca10056/nrf52840_pca10056.dts
```
152 arduino_serial: &uart1 {
153   status = "okay";
154   current-speed = <115200>;
155   rx-pin = <33>;
156   tx-pin = <34>;
157 };
158 
```

because uart1 is UARTE

how to reproduce:

enable UART_1_NRF_UARTE

```
west build -p auto  -b nrf52840_pca10056   samples/subsys/shell/shell_module
```

```
[0/1] Re-running CMake...
Zephyr version: 2.1.0
-- Selected BOARD nrf52840_pca10056
-- Found west: /home/aep/.local/bin/west (found suitable version "0.6.0", minimum required is "0.6.0")
-- Loading /home/aep/proj/artanis/os/zephyrproject/zephyr/boards/arm/nrf52840_pca10056/nrf52840_pca10056.dts as base
Devicetree configuration written to /home/aep/proj/artanis/os/zephyrproject/zephyr/build/zephyr/include/generated/generated_dts_board.conf
Parsing Kconfig tree in /home/aep/proj/artanis/os/zephyrproject/zephyr/samples/subsys/shell/shell_module/Kconfig
Loaded configuration '/home/aep/proj/artanis/os/zephyrproject/zephyr/build/zephyr/.config'
No change to '/home/aep/proj/artanis/os/zephyrproject/zephyr/build/zephyr/.config'
-- Cache files will be written to: /home/aep/.cache/zephyr
-- Configuring done
-- Generating done
-- Build files have been written to: /home/aep/proj/artanis/os/zephyrproject/zephyr/build
[126/143] Building C object zephyr/drivers/serial/CMakeFiles/drivers__serial.dir/uart_nrfx_uarte.c.obj
FAILED: zephyr/drivers/serial/CMakeFiles/drivers__serial.dir/uart_nrfx_uarte.c.obj 
ccache /opt/toolchains/gcc-arm-none-eabi-7-2018-q2-update/bin/arm-none-eabi-gcc -DBUILD_VERSION=v2.1.0-rc2-37-g92e017fd70b2 -DKERNEL -DNRF52840_XXAA -D_FORTIFY_SOURCE=2 -D__PROGRAM_START -D__ZEPHYR_SUPERVISOR__ -D__ZEPHYR__=1 -I../include -I../include/drivers -Izephyr/include/generated -I../soc/arm/nordic_nrf/nrf52 -I../ext/lib/fnmatch/. -I../ext/hal/cmsis/Core/Include -I/home/aep/proj/artanis/os/zephyrproject/modules/hal/nordic/nrfx -I/home/aep/proj/artanis/os/zephyrproject/modules/hal/nordic/nrfx/drivers/include -I/home/aep/proj/artanis/os/zephyrproject/modules/hal/nordic/nrfx/mdk -I/home/aep/proj/artanis/os/zephyrproject/modules/hal/nordic/. -I/home/aep/proj/artanis/os/zephyrproject/modules/debug/segger/rtt -isystem ../lib/libc/minimal/include -isystem /opt/toolchains/gcc-arm-none-eabi-7-2018-q2-update/bin/../lib/gcc/arm-none-eabi/7.3.1/include -isystem /opt/toolchains/gcc-arm-none-eabi-7-2018-q2-update/bin/../lib/gcc/arm-none-eabi/7.3.1/include-fixed -Os -imacros/home/aep/proj/artanis/os/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -mthumb -mcpu=cortex-m4 -imacros/home/aep/proj/artanis/os/zephyrproject/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -ffunction-sections -fdata-sections -mabi=aapcs -march=armv7e-m -std=c99 -nostdinc -MD -MT zephyr/drivers/serial/CMakeFiles/drivers__serial.dir/uart_nrfx_uarte.c.obj -MF zephyr/drivers/serial/CMakeFiles/drivers__serial.dir/uart_nrfx_uarte.c.obj.d -o zephyr/drivers/serial/CMakeFiles/drivers__serial.dir/uart_nrfx_uarte.c.obj   -c /home/aep/proj/artanis/os/zephyrproject/zephyr/drivers/serial/uart_nrfx_uarte.c
/home/aep/proj/artanis/os/zephyrproject/zephyr/drivers/serial/uart_nrfx_uarte.c:1386:15: error: 'DT_NORDIC_NRF_UARTE_UART_1_CURRENT_SPEED' undeclared here (not in a function); did you mean 'DT_NORDIC_NRF_UART_UART_1_CURRENT_SPEED'?
   .baudrate = DT_NORDIC_NRF_UARTE_UART_##idx##_CURRENT_SPEED,    \
               ^
/home/aep/proj/artanis/os/zephyrproject/zephyr/drivers/serial/uart_nrfx_uarte.c:1328:3: note: in expansion of macro 'UARTE_CONFIG'
   UARTE_CONFIG(idx),            \
   ^~~~~~~~~~~~
/home/aep/proj/artanis/os/zephyrproject/zephyr/drivers/serial/uart_nrfx_uarte.c:1440:2: note: in expansion of macro 'UART_NRF_UARTE_DEVICE'
  UART_NRF_UARTE_DEVICE(1);
  ^~~~~~~~~~~~~~~~~~~~~
/home/aep/proj/artanis/os/zephyrproject/zephyr/drivers/serial/uart_nrfx_uarte.c:1338:4: error: 'DT_NORDIC_NRF_UARTE_UART_1_BASE_ADDRESS' undeclared here (not in a function); did you mean 'DT_NORDIC_NRF_UART_UART_1_BASE_ADDRESS'?
    DT_NORDIC_NRF_UARTE_UART_##idx##_BASE_ADDRESS,        \
    ^
/home/aep/proj/artanis/os/zephyrproject/zephyr/drivers/serial/uart_nrfx_uarte.c:1440:2: note: in expansion of macro 'UART_NRF_UARTE_DEVICE'
  UART_NRF_UARTE_DEVICE(1);
  ^~~~~~~~~~~~~~~~~~~~~
/home/aep/proj/artanis/os/zephyrproject/zephyr/drivers/serial/uart_nrfx_uarte.c: In function 'uarte_1_init':
/home/aep/proj/artanis/os/zephyrproject/zephyr/drivers/serial/uart_nrfx_uarte.c:1349:15: error: 'DT_NORDIC_NRF_UARTE_UART_1_TX_PIN' undeclared (first use in this function); did you mean 'DT_NORDIC_NRF_UART_UART_1_TX_PIN'?
    .pseltxd = DT_NORDIC_NRF_UARTE_UART_##idx##_TX_PIN,    \
               ^
/home/aep/proj/artanis/os/zephyrproject/zephyr/drivers/serial/uart_nrfx_uarte.c:1440:2: note: in expansion of macro 'UART_NRF_UARTE_DEVICE'
  UART_NRF_UARTE_DEVICE(1);
  ^~~~~~~~~~~~~~~~~~~~~
/home/aep/proj/artanis/os/zephyrproject/zephyr/drivers/serial/uart_nrfx_uarte.c:1349:15: note: each undeclared identifier is reported only once for each function it appears in
    .pseltxd = DT_NORDIC_NRF_UARTE_UART_##idx##_TX_PIN,    \
               ^
/home/aep/proj/artanis/os/zephyrproject/zephyr/drivers/serial/uart_nrfx_uarte.c:1440:2: note: in expansion of macro 'UART_NRF_UARTE_DEVICE'
  UART_NRF_UARTE_DEVICE(1);
  ^~~~~~~~~~~~~~~~~~~~~
/home/aep/proj/artanis/os/zephyrproject/zephyr/drivers/serial/uart_nrfx_uarte.c:1350:15: error: 'DT_NORDIC_NRF_UARTE_UART_1_RX_PIN' undeclared (first use in this function); did you mean 'DT_NORDIC_NRF_UARTE_UART_1_TX_PIN'?
    .pselrxd = DT_NORDIC_NRF_UARTE_UART_##idx##_RX_PIN,    \
[... 56 lines cut ...]
    irq_enable(DT_NORDIC_NRF_UARTE_UART_##idx##_IRQ_0);), ())\
    ^~~~~~~~~~
/home/aep/proj/artanis/os/zephyrproject/zephyr/drivers/serial/uart_nrfx_uarte.c:1440:2: note: in expansion of macro 'UART_NRF_UARTE_DEVICE'
  UART_NRF_UARTE_DEVICE(1);
  ^~~~~~~~~~~~~~~~~~~~~
In file included from ../include/drivers/uart.h:26:0,
                 from /home/aep/proj/artanis/os/zephyrproject/zephyr/drivers/serial/uart_nrfx_uarte.c:11:
/home/aep/proj/artanis/os/zephyrproject/zephyr/drivers/serial/uart_nrfx_uarte.c: At top level:
/home/aep/proj/artanis/os/zephyrproject/zephyr/drivers/serial/uart_nrfx_uarte.c:1375:9: error: 'DT_NORDIC_NRF_UARTE_UART_1_LABEL' undeclared here (not in a function); did you mean 'DT_NORDIC_NRF_UART_UART_1_LABEL'?
         DT_NORDIC_NRF_UARTE_UART_##idx##_LABEL,         \
         ^
../include/device.h:107:11: note: in definition of macro 'DEVICE_AND_API_INIT'
   .name = drv_name, .init = (init_fn),     \
           ^~~~~~~~
/home/aep/proj/artanis/os/zephyrproject/zephyr/drivers/serial/uart_nrfx_uarte.c:1374:2: note: in expansion of macro 'DEVICE_DEFINE'
  DEVICE_DEFINE(uart_nrfx_uarte##idx,           \
  ^~~~~~~~~~~~~
/home/aep/proj/artanis/os/zephyrproject/zephyr/drivers/serial/uart_nrfx_uarte.c:1440:2: note: in expansion of macro 'UART_NRF_UARTE_DEVICE'
  UART_NRF_UARTE_DEVICE(1);
  ^~~~~~~~~~~~~~~~~~~~~
[135/143] Building C object zephyr/kernel/CMakeFiles/kernel.dir/timer.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: cannot run flash; no --build-dir given and build in current directory /home/aep/proj/artanis/os/zephyrproject/zephyr/build failed

```

anangl, 2019-12-04:
@aep 
I cannot reproduce this. The sample (`samples/subsys/shell/shell_module`) builds fine for me with `UART_1` enabled (I set `CONFIG_UART_1_NRF_UARTE=y` in this sample's `prj.conf`).

But looking at the console output you provided, I don't entirely get one thing (I divided the original line into three for better visibility):
```
/home/aep/proj/artanis/os/zephyrproject/zephyr/drivers/serial/uart_nrfx_uarte.c:1386:15:
error: 'DT_NORDIC_NRF_UARTE_UART_1_CURRENT_SPEED' undeclared here (not in a function);
did you mean 'DT_NORDIC_NRF_UART_UART_1_CURRENT_SPEED'?
```
Have you changed the `compatible` property in the `uart1` node to `"nordic,nrf-uart"`?
It should be `"nordic,nrf-uarte"` as originally set in [nrf52840.dtsi](https://github.com/zephyrproject-rtos/zephyr/blob/master/dts/arm/nordic/nrf52840.dtsi#L102). Then, the macro would be generated with the name `DT_NORDIC_NRF_UARTE_UART_1_CURRENT_SPEED`, as expected by the driver.

And what do you mean by:
> i suspect this dts is wrong in boards/arm/nrf52840_pca10056/nrf52840_pca10056.dts
> [..]
> because uart1 is UARTE
> 
> changing it to uarte makes it compile just fine.

What are you actually changing? There's no need to correct anything here. As I wrote above, the `compatible` property for the `uart1` node should be set to `"nordic,nrf-uarte"`, and it is so by default.

aep, 2019-12-04:
well,... embarrassing. Apparently i broke this. it works fine with a fresh clone and going through my own repro steps. sorry :/

closed 2019-12-04 by aep

anangl, 2019-12-04:
> well,... embarrassing. Apparently i broke this. it works fine with a fresh clone and going through my own repro steps. sorry :/

No worries. It happens. Good that we managed to solve this quickly and now it works for you.