zephyrproject-rtos/zephyr #21155: Port drivers to use sys/ring_buffer

ceolin, 2019-12-03:
There are some drivers that are implementing their own ring_buffer, there are cases where even functions names are colliding with sys/ring_buffer.

Drivers doing it:
`drivers/ethernet/eth_sam_gmac.c`
`drivers/i2s/i2s_sam_ssc.c`
`drivers/i2s/i2s_ll_stm32.h`

nandojve, 2021-07-03:
CC: @mnkp

nashif, 2021-09-29:
to test this, you will need a platform that uses the above drivers..

wolterhv, 2022-01-21:
So, to solve this issue, we need to be able to test it (at least)

- on a MCU from the Atmel SAM family which supports ethernet and I2S and
- on a MCU from the STM32 family which supports I2S.

It seems to me however that this would be a final validation test. In the meantime, I don't see why the replacement of the ring buffer couldn't be unit tested to verify functional equivalence with the current implementation. What do you think about this @nashif ?

Spice-Weasel, 2023-10-27:
Think I've got a couple ATSAM boards that I can try this on, not supported devkits though.

nandojve, 2023-11-01:
I can help with network test for SAM. I don't have a way to test SSC driver but if it uses the same approach from networking I think solution will be similar and we can move forward to replace for the SAM's.

noobninjanerd, 2025-10-30:
Hello, I would like to try and take a crack at this - the STM32 part. I have a Nucleo-F410RB and it is I2S capable.

I checked `drivers/i2s/i2s_ll_stm32.h` and it doesn't seem to have a local ring_buffer implementation, just a pointer queue. I guess I could incorporate zephyr's `sys/ring_buffer` into it.

`drivers/i2s/i2s_sam_ssc.c` does have both the queue and the ring_buffer so I can refactor that and at the least unit test it.

I am very new to this so feel free to give suggestions and point me to any docs that could be helpful.