zephyrproject-rtos/zephyr #21156: Interrupts do not work on UP Squared board

dcpleung, 2019-12-03:
**Describe the bug**
`samples/boards/up_squared/gpio_counter` no longer works on UP Squared due to interrupts no longer being delivered.

**To Reproduce**
Steps to reproduce the behavior:
1. cd samples/boards/up_squared/gpio_counter/
2. mkdir build; cd build
3. cmake -DBOARD=up_squared
4. make
5. Set BIOS configurations described inside `src/main.c`.
6. Boot the Zephyr image according to documentation.
7. Observe the counter would not increment when inputs are pulsed on HAT pin 16.

**Expected behavior**
The counter would increment, and reflected on the 4-bit output of pin35/37/38/40.

**Impact**
- Broken board support.
- Blocking verification of GPIO API changes and app updates.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.10.3
 - Commit: 58e2057aaf9af4d299814f4aece585e287587162

dleach02, 2019-12-04:
Change was reverted.

jhedberg, 2020-02-12:
@dcpleung do you know if anyone is looking at this? Should it be reassigned to someone else?

dcpleung, 2020-02-12:
I have been looking at this on-and-off. The problems are lack of documentation on IO-APIC, and QEMU having the 0xFF broadcast which does not seem to be the case for actual hardware. One of the solution is to use logical destination, which works on UP^2 but fails QEMU. So more investigation is needed.

andrewboie, 2020-02-19:
Reopening this. I had to revert @ZideChen0 change #22848 as it was causing timeouts when running tests on qemu_x86_64. Needs further analysis.

dcpleung, 2020-03-30:
Another attempt #23325.

PR #21087 (2019-12-03): interrupt_controller: ioapic_intr: revert CPU target change
Commit 5a9a33b0cf943312dfe4de6e4525087601a741e4 changes interrupt
destination in an attempt to broadcast interrupts. However, this
change causes interrupts to stop working on the UP Squared board
in non-SMP configuration. According to QEMU source code,
physical destination address 0xFF000000 is a special case where
it broadcasts the interrupts. However, none of the IOAPIC
documentation (that I can find) describes this behavior. So,
revert that commit.

Fixes #21156
PR #22848 (2020-02-19): interrupt_controller: fixed IO APIC broadcasting issue
Currently IO APIC is working in physical mode which is not good for SMP.

Tested with samples/subsys/console/echo, and checked if it can receive keyboard inputs. Results:

without these patches (head: bf3850c27092)
           acrn  qemu_x86   qemu_x86_64(1CPU -- 2 CPUs)  up_sqaured(1CPU -- 2 CPUs)
xapic       N       Y                                       Y               N                                Y            Y
x2apic     N       ?                                        ?               ?                                N             [... cut]
PR #23325 (2020-05-09): interrupt_controller: fixed IO APIC broadcasting issue (V2)
This is the updated version for this PR which was closed:
https://github.com/zephyrproject-rtos/zephyr/pull/22848

Currently IO APIC is working in physical destination mode, fixed delivery, and the destination fields in  IOREDTBL is fixed to 0xff, which seems a invalid configuration. By definition, in this setting, IO APIC interrupt can be delivered to one single local APIC whose LAPIC ID is 0xff.

This patchset is intended to fix the IO APIC delivery issue on SMP systems:

- change IO APIC from physical mode to logical destination mo [... cut]
closed 2020-05-09 by nashif