zephyrproject-rtos/zephyr #21161: question: openthread with other boards

YC-Chiu, 2019-12-04:
Hello everyone!
I have read that - 
OpenThread becomes functional and integrates with all of the platforms Zephyr supports.
(https://www.prnewswire.com/news-releases/the-zephyr-project-announces-openthread-as-the-first-thread-protocol-implementation-to-integrate-with-zephyr-rtos-300604447.html)

Can I use other platforms? I want to use echo_server/client with my host boards that are not arm-based and other ieee802154 devices that are not arm-based too. Should I follow https://openthread.io/guides/porting to port a new platform first and add my device driver?

rlubos, 2019-12-04:
Hi @Y-U-KAY,

The guide you've posted refers to bare-metal OpenThread applications and does not apply in Zephyr. The OpenThread platform abstraction mentioned in the guide is already implemented for Zephyr, and is built on top of several Zephyr APIs (https://github.com/zephyrproject-rtos/zephyr/tree/master/subsys/net/lib/openthread/platform).

In order to run OpenThread on your platform, you need to make sure that it supports all of the Zephyr drivers required by OpenThread, which currently are `timer`, `flash`, `entropy` and of course `ieee802154` radio driver. Optionally, you might also need `uart` or any other shell backend for OpenThread shell.

Y-U-KAY, 2019-12-04:
@rlubos Thank you for your responses!

Is it not necessary to use the devices that have been ported in the OpenThread repository? 
Openthread required `flash` for firmware image. If I don't use the devices that have been ported in the OpenThread repository, is `flash` necessary?

jukkar, 2020-01-28:
Old question which was answered -> closing.

closed 2020-01-28 by jukkar