zephyrproject-rtos/zephyr #21162: Sanitycheck corrupted test case names in test-report.xml files

maksimmasalski, 2019-12-04:
**Describe the bug**
Sometimes during the Sanitycheck I receive test results .xml files with the corrupted test case names. Inside of .xml file among test cases with proper names, can be detected names like that one below.
`sys.mutex.nouser.thread_06@119. Failed to take mutex 0x20002288`
Error in that test case was

=======================================================
***** Booting Zephyr OS build zephyr-v2.0.0-1958-g53f30bc3ac24 *****
Running test suite mutex_complex

starting test - test_mutex
starting test - Test kernel Mutex API

Done LOCKING! Current priority = 5

Assertion failed at ZEPHYR_BASE/tests/kernel/mutex/sys_mutex/src/main.c:306: test_mutex: (rv not equal to 6)

thread_05 timed out and out priority should drop.
FAIL - test_mutex

starting test - test_user_access
SKIP - test_user_access

starting test - test_supervisor_access
PASS - test_supervisor_access
Test suite mutex_complex failed.

PROJECT EXECUTION FAILED

FAIL - thread_06@119. Failed to take mutex 0x20002288

================================================================

**To Reproduce**
1. It was a sanitycheck run for the board qemu_cortex_m3 during the master build master-2019-11-10-53f30bc3ac
2. I was not able to reproduce that bug again using the same commit 53f30bc3ac244417384b62e23e577a04a9687926
and running the sanitycheck for that test case and the board using `$sanitycheck -p qemu_cortex_m3 -T /home/maksim/zephyrproject/zephyr/tests/kernel/mutex/sys_mutex/`
That bug is difficult to catch.
3. Original output test-report.xml file with that error
you can find there https://github.com/maksimmasalski/share-files

**Additional comments**
1. Recently on November 29th the same error happened with the net.trickle for board mimxrt1050_evk.
`net.trickle.Trickle 1 0x80000c88 callback called`
The most recent commit with that error was 42c5b0a7fa
https://zephyrproject.testrail.io/index.php?/tests/view/14520629&group_by=cases:title&group_order=asc&group_id=-1
I can't reproduce for that commit that error too using `$sanitycheck -p mimxrt1050_evk -T /home/maksim/zephyrproject/zephyr/tests/net`
When I ran sanitycheck, that test passed successfully. Original output test-report.xml file with that error you can find there https://github.com/maksimmasalski/share-files

2. The same errors happened with the sys.mutex tests:
`sys.mutex.nouser.thread_06@119. Failed to take mutex 0x0012e088`
`sys.mutex.thread_06@119. Failed to take mutex 0x00110060`

You can try to find more corrupted test case names in the TestRail for the Zephyr RTOS 
https://zephyrproject.testrail.io

**Expected behavior**
In normal conditions I expect even if test case fails to pass, error message in the Sanitycheck can't destroy test case name string in the test-report.xml file. Many other test cases also fail to pass, but their test case names not corrupted with the error message. 

**Environment:**
 - OS: Fedora 28
 - SDK Zephyr 0.10.3
 - 53f30bc3ac244417384b62e23e577a04a9687926 or 42c5b0a7fa

maksimmasalski, 2019-12-04:
Today I will check to run full sanitycheck cycle for that boards, to see if the error can be reproduced.

nashif, 2019-12-05:
moving up to medium, this bug is causing sanitycheck to produce inconsistent test results when something fails.

maksimmasalski, 2019-12-05:
> moving up to medium, this bug is causing sanitycheck to produce inconsistent test results when something fails.

I'm in process of checking your fix #21191 , I need time to verify it, to understand if it will help on 100%.

PR #21191 (2019-12-09): sanitycheck: parse testcase names correctly
We were parsing random FAIL messages from the output of test runs ad
testcases and capturing them in the xml output. Now we only parse the
name if it starts with test_.

Fixes #21162
commit a4dd49b3 (2019-12-09): sanitycheck: parse testcase names correctly

We were parsing random FAIL messages from the output of test runs ad
testcases and capturing them in the xml output. Now we only parse the
name if it starts with test_.

Fixes #21162
closed 2019-12-09 by nashif