zephyrproject-rtos/zephyr #21165: Bluetooth: Mesh: Friend Clear message from a Friend node

Campou, 2019-12-04:
In Zephyr's implementation, when a Friend node tries to send the Friend Clear message to another Friend node, it will use the master security material from `bt_mesh.sub[0]`, and the code is:
```
static void send_friend_clear(struct bt_mesh_friend *frnd)
{
	struct bt_mesh_msg_ctx ctx = {
		.net_idx  = frnd->net_idx,
		.app_idx  = BT_MESH_KEY_UNUSED,
		.addr     = frnd->clear.frnd,
		.send_ttl = BT_MESH_TTL_MAX,
	};
	struct bt_mesh_net_tx tx = {
		.sub  = &bt_mesh.sub[0],
		.ctx  = &ctx,
		.src  = bt_mesh_primary_addr(),
		.xmit = bt_mesh_net_transmit_get(),
	};
        ......
```
Here, is there any chance that the other Friend node is not on the same subnet?

Thanks.

aescolar, 2019-12-04:
CC @jhedberg

jhedberg, 2019-12-04:
Thanks for the report @Campou. I think the above code is not quite right - it should probably be `.sub = bt_mesh_subnet_get(frnd->net_idx)`.

Campou, 2019-12-05:
Hi @jhedberg 

Thanks for the reply. And one more question here:

When LPN tries to send Friend Request to all-friend nodes, it will use the NetKey from `bt_mesh.sub[0]`, is it possible here that users can choose a different NetKey for the Friend Request, i.e. maybe using the `net_idx` as a parameter when sending the Friend Request?

BR

jhedberg, 2019-12-05:
@Campou currently, to keep the code simple, the LPN implementation only supports establishing Friendship on the primary subnet. If you have a requirement to support Friendship on multiple subnets you are welcome to submit a PR. I'm not sure it makes sense for an LPN to establish Friendship on only part of its subnets, since it would be unable to operate using low power otherwise. Even having Friendship on multiple subnets will consume substantially more power (more timers, PDUs to send, etc), which is why I made the assumption most LPNs will be happy being on a single subnet.

Btw, are you planning to submit a PR for the first issue you reported with the Friend Clear?

Campou, 2019-12-05:
Hi @jhedberg 

Thanks, I will submit a PR for the first issue.

PR #21204 (2019-12-09): Bluetooth: Mesh: Fixes wrong subnet used for Friend Clear
When Friend node tries to send Friend Clear message to other
Friend nodes, it should use the subnet information based on
the net_idx from friendship.

Fixes #21165
commit 2f691d25 (2019-12-09): Bluetooth: Mesh: Fixes wrong subnet used for Friend Clear

When Friend node tries to send Friend Clear message to other
Friend nodes, it should use the subnet information based on
the net_idx from friendship.

Fixes #21165
closed 2019-12-09 by nashif