zephyrproject-rtos/zephyr #21171: Module Request: Optiga Trust X

panmasuo, 2019-12-04:
Request for repo for Optiga Trust X security chip. 

Name of repo: optiga-trust-x

[Optiga Trust X framework](https://github.com/Infineon/optiga-trust-x)
[Discussion on mailing list](https://lists.zephyrproject.org/g/devel/topic/add_support_optiga_trust_x/54228013)

I managed to port this library with Zephyr's mbedTLS and exclude "internal" mbedTLS to avoid redundancy with library.
All examples included in *examples/optiga* and *example_authenticate_chip.c* tested and working on efm32pg_stk3402a.

MarcDorner, 2019-12-04:
Hi,
I am currently also working on a port of the [OPTIGA Trust M Framework](https://github.com/Infineon/optiga-trust-m) for Zephyr. Right now my port is not within this repository but will follow in the next weeks.

Since @panmasuo's work and mine are quite similiar but are targeted at different OPTIGA Trust chips/frameworks, I had a structure like the following in mind within the Zephyr Project repository: 
* `optiga_trust/`
    * `optiga-trust-x/`
    * `optiga-trust-m/`

The basic idea is to have a main `optiga_trust` repository within the Zephyr project, where each framework code would be put in a subfolder. That way all OPTIGA Trust code would be at a unified location.

I also had a brief talk with @carlescufi about this idea.

carlescufi, 2019-12-05:
> Hi,
> I am currently also working on a port of the [OPTIGA Trust M Framework](https://github.com/Infineon/optiga-trust-m) for Zephyr. Right now my port is not within this repository but will follow in [...]

I think your proposal makes a lot of sense @MarcDorner . I have created the repo: https://github.com/zephyrproject-rtos/optiga_trust

I suggest you place this in `modules/crypto` perhaps? I think it's better than `modules/hal`.

carlescufi, 2019-12-05:
@panmasuo see above thread, repo created.

closed 2019-12-05 by carlescufi