zephyrproject-rtos/zephyr #21179: Reduce RAM consumption for civetweb HTTP sample

erwango, 2019-12-04:
**Is your enhancement proposal related to a problem? Please describe.**
Current configuration for Civetweb HTTP sample requires too much RAM (>230K), which prevents it from being run on wide number of targets.

This choice was deliberate at PR introduction (https://github.com/zephyrproject-rtos/zephyr/pull/17019) as described in this [comment](https://github.com/zephyrproject-rtos/zephyr/pull/17019#issuecomment-505387159).

Civetweb is the only HTTP client sample available today for zephyr. This memory consumption makes is unavailable for many targets, so situation  is like there were no Zephyr HTTP client for many Zephyr users.

Before net refactoring, I was able to run a micropython powered HTTP server on top of existing networking stack on nucleo_f429zi (192K RAM), so I would expect we could have a zephyr native alternative running on the same platform today.

`samples/net/sockets/civetweb` configuration shows that main factor for RAM consumption is the following setting:
```
CONFIG_MINIMAL_LIBC_MALLOC_ARENA_SIZE=131072
```
Below 92000 an illegal exception appears. At 92000 connection closes after a few history clicks.
So getting it down to an acceptable level obviously requires some fine tuning and potentially further optimisation.

**Describe the solution you'd like**
Rework civetweb sample configuration to fit below an acceptable RAM consumption.

**Describe alternatives you've considered**

**Additional context**

jukkar, 2019-12-04:
> Civetweb is the only HTTP client sample available today for zephyr. This memory consumption makes is unavailable for many targets, so situation is like there were no Zephyr HTTP client for many Zephyr users.

This is not correct atm. There is a light weight HTTP client API that is currently used by the Websocket client library. The API can be found in `include/net/http_client.h` and `include/net/websocket.h` header files, the code can be found at `subsys/net/lib/http` and `subsys/net/lib/websocket` directories.

erwango, 2019-12-04:
> This is not correct atm. There is a light weight HTTP client API that is currently used by the Websocket client library. The API can be found in include/net/http_client.h and include/net/websocket.h header files, the code can be found at subsys/net/lib/http and subsys/net/lib/websocket directories.

Thanks, I have missed that. I'll have a look, tough the concern for civetweb is still valid (but less important maybe).

jukkar, 2019-12-05:
Lowering the priority as there is alternative implementation available for HTTP client side. For server side HTTP support there is nothing else available atm.

commit c28ffbb2 (2020-09-30): samples: net: civetweb: reduce RAM usage

adjust configuration so that civetweb uses less memory
fixes #21179
closed 2020-09-30 by jukkar