zephyrproject-rtos/zephyr #21187: Can not ping or run http server via ethernet when gPTP is enabled

bcdonovan, 2019-12-04:
**Describe the bug**
I have a small application which combines the gPTP example and the dumb_http_server example. I have merged both code and the respective prf.conf entries. 

I can run gPTP or the http server but not both. 

When gPTP is enabled (via CONFIG_NET_GPTP=y) then I can not ping the device and the http server does not respond. This is true with or without a gPTP grand master operating (AVnu gptp). With gPTP enabled I am able to sync using AVnu gptp.

When gPTP is disabled then I can both ping and the http server will respond to the client. 

I have attempted to adjust thread priorities and the scheduler algorithm and this does not appear to change the symptoms. 

**Expected behavior**
I would like to be able to run gPTP while running IPv4 communications

**Impact**
Prevents adoption of Zephyr

**Screenshots or console output**
N/A

**Environment (please complete the following information):**
This is on a NXP MIMRT1064-EVK board.
Ubuntu 19.10
zephyr git tag: v2.1.0-rc3
nxp_hal branch: master or v1.14-branch
zephr sdk: 0.10.3

**Additional context**
Note: Had to fix an infinite loop issue in nxp_hal to get gPTP to work on the nxp_hal master branch. The v1.14-branch of nxp_hal worked unmodified. Patch for nxp_hal master branch:

```
diff -r nxp/mcux/drivers/imx/fsl_enet.c nxp_master/mcux/drivers/imx/fsl_enet.c
9,11d8
< #include <logging/log.h>
< LOG_MODULE_REGISTER(enet_dbg, LOG_LEVEL_DBG);
< 
2910,2914d2906
<         }
< 
<         /* check to see if next buffer is the same as the last if so break */
<         if (handle->txBdDirtyTime[ringId] == curBuffDescrip) {
<             break;
```

jukkar, 2019-12-04:
Interesting issue. There should not be anything in networking stack that would prevent IP protocols from working if gPTP is enabled. I will try this with my TSN setup tomorrow when I have access to my test network.

jukkar, 2019-12-05:
Tested this in `sam-e70` with echo-server and gptp, and did not see any issues with IP connectivity. I was not able to test the setup with `frdm-k64f` or `mimxrt1050_evk` because the console was not working and network interface was always down, dunno what is wrong there (this was with latest upstream).

I tested this scenario like this:
```
$ cd samples/net/sockets/echo_server
$ cmake -B build -DBOARD=sam_e70_xplained . -DCONFIG_WDT_DISABLE_AT_BOOT=y -DCONFIG_NET_GPTP=y -DCONFIG_NET_GPTP_STATISTICS=y
$ make -C build flash
```
Then from host terminal, I launched echo-client like this
```
$ cd net-tools
$ ./echo-client 2001:db8::1 -t
.......
Average round trip 29 ms
Sent 7 packets
```
Also ping works ok
```
$ ping 2001:db8::1
PING 2001:db8::1(2001:db8::1) 56 data bytes
64 bytes from 2001:db8::1: icmp_seq=1 ttl=64 time=0.304 ms
64 bytes from 2001:db8::1: icmp_seq=2 ttl=64 time=0.298 ms
64 bytes from 2001:db8::1: icmp_seq=3 ttl=64 time=0.299 ms
^C
--- 2001:db8::1 ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 34ms
rtt min/avg/max/mdev = 0.298/0.300/0.304/0.014 ms
```
Tested gPTP against OpenAVNU gPTP daemon and time sync worked just fine.
```
uart:~$ net gptp 1
Port id    : 1
Clock id   : fc:c2:3d:ff:fe:11:c1:47
Version    : 2
AS capable : yes
...
Statistics:
Sync messages received                 : 5333
Follow Up messages received            : 6030
Path Delay Request messages received   : 699
Path Delay Response messages received  : 697
Path Delay messages threshold exceeded : 0
Path Delay Follow Up messages received : 697
Announce messages received             : 698
ptp messages discarded                 : 0
Sync reception timeout                 : 698
Announce reception timeout             : 0
Path Delay Requests without a response : 368
Sync messages sent                     : 0
Follow Up messages sent                : 0
Path Delay Request messages sent       : 1065
Path Delay Response messages sent      : 699
Path Delay Response FUP messages sent  : 699
Announce messages sent                 : 699
```

agansari, 2019-12-05:
@jukka thank you for testing, what you say it's eth_mcux issue and not a networking stack problem.

jukkar, 2019-12-05:
> eth_mcux issue and not a networking stack problem.

Well, it certainly looks like that.

bcdonovan, 2019-12-05:
I've repeated portions of the test @jukkar ran and am seeing different behavior between IPv4 and IPv6. In short:

gPTP enabled: IPv6 works (both ping and echo), IPv4 fails (both ping and echo)
gPTP disabled: IPv6 works (both ping and echo), IPv4 works (both ping and echo)

I added 

```
CONFIG_WDT_DISABLE_AT_BOOT=y 
CONFIG_NET_GPTP=y 
CONFIG_NET_GPTP_STATISTICS=y
```

to my prof.conf

and then ran

```
west build -p auto -b mimxrt1064_evk samples/net/sockets/echo_server
```

and flashed using NXP's `crt_emu_redlink`

My test setup is the NXP dev board connected directly to an ubuntu laptop via ethernet.

jukkar, 2019-12-05:
In my examples above, I had only IPv6 logs printed but I did test the IPv4 functionality too and there was no difference between them.

bcdonovan, 2019-12-09:
The issue appears to be due to IP accelerators on the NXP 1064 processor. Using gPTP is enabling additional 1588 hardware on the NXP processor. This appears to be enabling a hardware accelerator which will calculate and insert the IP header checksum. The hardware IP accelerator requires that the IPv4 header checksum from the software application be set to 0 otherwise bad IPv4 header values are showing up in wireshark. I have not found any documentation stating that the checksum accelerator is enabled when the 1588 hardware is enabled but this is the behavior that I am seeing. 

I've been able to get gPTP, ping and http to work if I 0 the `ipv4_hdr->chksum` value in:

`zephyr/subsys/net/ip/ipv4.c` -> function `net_ipv4_finalize`

jukkar, 2019-12-10:
Zephyr supports checksum offloading for RX and TX network packets so if the Ethernet drivers sets `ETHERNET_HW_TX_CHKSUM_OFFLOAD` capability, then the IP stack will not calculate the checksum for TX packets. Unfortunately atm this setting applies to IPv4, TCP and UDP protocols so if NXP hw only calculates it for IPv4, then this setting cannot be used.

agansari, 2019-12-10:
@bcdonovan please test pull #21269 it enables hw checksum on bot rx and tx.
I tested on i.mx 1050 rt, echo_server with -DCONFIG_NET_GPTP=y -DCONFIG_NET_GPTP_STATISTICS=y, pinging works on both ipv4 and ipv6.

Still having issues with K64F which uses the same driver when enabling gptp.

agansari, 2019-12-10:
@jukkar @bcdonovan by checking the registers, checksum was disabled

agansari, 2019-12-10:
k64f is slower, tx buffers are flooded, increasing Tx bufers number fixes the issue, i.MX has a much faster core.

```
 config ETH_MCUX_TX_BUFFERS
 	int "Number of MCUX TX buffers"
-	default 1
-	range 1 1
+	default 4
+	range 1 16
```

pfalcon, 2019-12-16:
> please test pull #21269 it enables hw checksum on both rx and tx.

So, I'm testing #21269, and as it is, it visibly regresses frdm_k64f. While there seems to be a way to "fix" it, I personally would have reservation about enabling acceleration features by default. Reason should be obvious: with software issues, we can debug and fix them, with hardware issues, debugging them is much more complicated and fixing are oftentimes not possible. Actually, even figuring out that hardware accel may be involved with some issue may take quite some time and pulled hair. (I'd say this ticket rather shows that, than not.)

And reading thru comments in this ticket, I don't see immediate connection between what's discussed here and #21269. I applying some *guesswork*, I may imagine the logic *might* be: "If there's a speculation (not backed by doc references so far, not backed by checking hardware state) that gPTP enables hardware accel, let's just enable it by default". But I wouldn't like to do guesswork, so @agansari, could you please explain in detail the relation of this issue and that patch?

And even assuming the above somehow mirrors the actual situation, I'd personally stay on the safe side, and: a) would make eth_mcux hw accel to be a Kconfig option; b) keep it off by default like it was before, and only enable it in case of gPTP is enabled.

agansari, 2019-12-19:
@pfalcon after some code (fsl_enet), documentation investigation (i.mx 1050 and k64f ENET chapter) and configuration testing i found the following:
 - the issue is observed on Tx frames
 - the software culprit is ENET_ENHANCEDBUFFERDESCRIPTOR_MODE
 - hardware wise, the definition above sets hardware in a new state as you _guesstimate_.

So, gPTP need an extended buffer descriptor to store 1588 timestamps, and this is okay; the issue we have is registers PINS and IINS are not used in software. Extended buffers aren't used in 'normal' mode, a legacy buffer descriptor is used and checksum is correctly passed to networking hardware.

Enabling overall hw accelerated ip+protocol checksum makes it work correctly, and I find enabling this for each buffer descriptor is just an overhead.
I advocate for enabling default hardware acceleration and will add my comment in #21269.

Reference: (i.mx1050/k64f) 24.6.14.2/45.5.14.2 Enhanced transmit buffer descriptor. 

**Also, we need >=2 Tx buffers in this scenario**, gptp floods the driver and ARP can't be reached.

jukkar, 2019-12-30:
Re-opening as there are issues still with frdm-k64f

PR #21269 (2019-12-20): drivers: eth_mcux enable checksum and autonegotiation
Enable ETHERNET_HW_TX_CHKSUM_OFFLOAD, ETHERNET_HW_RX_CHKSUM_OFFLOAD,
ETHERNET_AUTO_NEGOTIATION_SET and the equivalent driver configuration in
eth_mcux driver. Autonegotiation was done at driver initialization.

Fixes #21187
closed 2020-08-28 by github-actions[bot]