zephyrproject-rtos/zephyr #21201: ARM: Core Stack Improvements/Bug fixes for 2.2 release

ioannisg, 2019-12-05:
- [ ] [Bug ARM, Kernel] Make threads use `K_THREAD_STACK_SIZEOF `and remove workaround for ARM 's `z_new_thread()` (#14269)
- [ ] [Bug ARM Cortex-M] Support for HW stack protection features together with CONFIG_ARM_SECURE_FIRMWARE=y and CONFIG_FP_SHARING (currently broken)
- [ ] [Enh. ARCHitectures, Kernel] Support distinct ways for allocating stack areas for default and supervisor-only threads (#13637)
- [ ] [Enh. ARM] Add HW stack protection during boot, too.
- [ ] [Enh. ARMv8-m] Make built-in stack overflow a default feature (=y)
- [ ] [ARM Cortex-M, ARC, Kernel] Consider moving the arm_core_mpu API above the arch- level, so as not to duplicate for ARM, ARC
- [ ] [Enh. Kernel] Rename `userlocal_data `to something that does not contain "user" in the name, as it is used under `CONFIG_USERSPACE ` but beyond user threads
- [ ] [Enh. ARM] Document the incompatibilities between ARMv7-M and ARMv8-M Mainline
- [ ] [Enh. ARM Cortex-M] Add ARM System Call test

jhedberg, 2020-02-04:
@ioannisg removed this from the release plan since it doesn't look like it'll make it. It didn't feel right to move it to 2.3 either since some items have been completed.

ioannisg, 2020-02-10:
> @ioannisg removed this from the release plan since it doesn't look like it'll make it. It didn't feel right to move it to 2.3 either since some items have been completed.

@jhedberg that's fine, so, what I will do is to re-scope it and close this before the release, and move the undone tasks to next release.

ioannisg, 2020-03-10:
Closing this meta-issue; unfinished tasks moved to #23366

closed 2020-03-10 by ioannisg