zephyrproject-rtos/zephyr #21202: Required upgrade of HAL

pfl, 2019-12-05:
After upgrading, something like the following starts to happen suddenly:
```
In file included from /home/pflykt/projects/zephyrproject/modules/hal/nordic/nrfx/mdk/nrf.h:124,
                 from /home/pflykt/projects/zephyrproject/modules/hal/nordic/nrfx/drivers/nrfx_common.h:39,
                 from /home/pflykt/projects/zephyrproject/modules/hal/nordic/nrfx/nrfx.h:36,
                 from ../../../soc/arm/nordic_nrf/nrf52/soc.h:16,
                 from /home/pflykt/projects/zephyrproject/zephyr/drivers/clock_control/nrf_power_clock.c:8:
/home/pflykt/projects/zephyrproject/zephyr/drivers/clock_control/nrf_power_clock.c: In function 'clock_event_check_and_clean':
/home/pflykt/projects/zephyrproject/modules/hal/nordic/nrfx/mdk/nrf52840.h:2826:37: error: incompatible type for argument 1 of 'nrf_clock_event_check'
 #define NRF_CLOCK                   ((NRF_CLOCK_Type*)         NRF_CLOCK_BASE)
                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/pflykt/projects/zephyrproject/zephyr/drivers/clock_control/nrf_power_clock.c:53:35: note: in expansion of macro 'NRF_CLOCK'
  bool ret = nrf_clock_event_check(NRF_CLOCK, evt) &&
```
After a bit of investigation, the above and other weird compile errors mean that the board HAL needs to be updated. 

A much better usability solution would be to detect the HAL version in use and print out a warning that the installed and expected HAL versions differ.

carlescufi, 2019-12-05:
@pfl just checking, have you run `west update`?

carlescufi, 2019-12-05:
@pfl I assume you have run `west update` and that this issue covers the fact of handling this in a nicer way.

jukkar, 2019-12-05:
I think the idea was that west/cmake could check that modules are according to what manifest says and possibly give some information to the user if there is a mismatch so that user would know that he needs to run "west update".

pfl, 2019-12-09:
> @pfl I assume you have run `west update` and that this issue covers the fact of handling this in a nicer way.

Yes, I did run west update as a result. So this is about either documenting such a situation, or, as @jukkar commented, have west/cmake notice that the HAL is too old wrt to the code being compiled. The latter approach would be preferred.

carlescufi, 2019-12-09:
Since this issue only covers HAL repos but the problem is in fact more general and already described here: https://github.com/zephyrproject-rtos/west/issues/333 I will close this issue in benefit of the one linked.

closed 2019-12-09 by carlescufi