zephyrproject-rtos/zephyr #21205: get_device_list only available if power management invoked

WilliamGFish, 2019-12-05:
**Issue**
Can the get_device_list function in device.c be always available? Why would it only be valid if power management functionality invoked?

**Solution**
Remove the constitutionality from these functions

code snippet
```
#ifdef CONFIG_DEVICE_POWER_MANAGEMENT
int device_pm_control_nop(struct device *unused_device,
		       u32_t unused_ctrl_command,
		       void *unused_context,
		       device_pm_cb cb,
		       void *unused_arg)
{
	return 0;
}

void device_list_get(struct device **device_list, int *device_count)
{

	*device_list = __device_init_start;
	*device_count = __device_init_end - __device_init_start;
}
```
**Additional context**
This will be useful in sensor configuration development.

ceolin, 2021-06-04:
This function was removed and `z_device_get_all_static` should be used.

closed 2021-06-04 by ceolin

WilliamGFish, 2021-06-07:
> This function was removed and `z_device_get_all_static` should be used.

Am I able to use this function without implementing Power Management?

Billy..