zephyrproject-rtos/zephyr #21229: cc1plus: warning: '-Werror=' argument '-Werror=implicit-int' is not valid for C++

adfernandes, 2019-12-06:
I recently started testing the `gcc-arm-none-eabi-9-2019-q4-major` compiler in a C++ project, and came across the following for my C++ translation units:
```
cc1plus: warning: '-Werror=' argument '-Werror=implicit-int' is not valid for C++
```

This is a known change in `gcc 9` and will likely is or will soon be a part of `clang`.

**To Reproduce**
Enable C++ and add a C++ file to the project.

**Expected behavior**
The `-Werror=implicit-int` option should not be enabled for C++ translation units.

**Impact**
This is an annoyance because Zephyr appears to really pride itself on "zero warning" builds, which is expected of possible mission- or safety-critical embedded systems.

**Environment**
 - macOS and Linux
 - `gcc-arm-none-eabi-9-2019-q4-major`
 - Zephyr `v2.1.0-rc3-6-gf570f19bef`

**First Steps Fixing**

_I'm volunteering to help fix this, if some kind person can give me some pointers!_

The macro `toolchain_cc_warning_error_implicit_int` is defined
* [here](https://github.com/zephyrproject-rtos/zephyr/blob/deb7ff524a73216c9907b8b6ef61efc27341b789/cmake/compiler/gcc/target_warnings.cmake#L91) for `gcc`, and 
* [here](https://github.com/zephyrproject-rtos/zephyr/blob/4a5751226c40d56644f8062ef568562d3db07f36/cmake/compiler/clang/target_warnings.cmake#L98) for `clang`

The macro `toolchain_cc_warning_error_implicit_int` is **enabled** [here](https://github.com/zephyrproject-rtos/zephyr/blob/d58f8bee79dddb9aa1b65a2ef1624ebf1b389a4e/CMakeLists.txt#L290)

Now, following the chain:
* [`toolchain_cc_warning_error_implicit_int`](https://github.com/zephyrproject-rtos/zephyr/blob/deb7ff524a73216c9907b8b6ef61efc27341b789/cmake/compiler/gcc/target_warnings.cmake#L91) calls [`zephyr_cc_option`](https://github.com/zephyrproject-rtos/zephyr/blob/afe6dab155000386f4adeadf20b65f14125e4218/cmake/extensions.cmake#L116)
* [`zephyr_cc_option`](https://github.com/zephyrproject-rtos/zephyr/blob/afe6dab155000386f4adeadf20b65f14125e4218/cmake/extensions.cmake#L116) calls [`target_cc_option`](https://github.com/zephyrproject-rtos/zephyr/blob/afe6dab155000386f4adeadf20b65f14125e4218/cmake/extensions.cmake#L1310)
* [`target_cc_option`](https://github.com/zephyrproject-rtos/zephyr/blob/afe6dab155000386f4adeadf20b65f14125e4218/cmake/extensions.cmake#L1310) calls [`target_cc_option_fallback`](https://github.com/zephyrproject-rtos/zephyr/blob/afe6dab155000386f4adeadf20b65f14125e4218/cmake/extensions.cmake#L1316)

**Now the problem is (I think)** that [`target_cc_option_fallback`](https://github.com/zephyrproject-rtos/zephyr/blob/afe6dab155000386f4adeadf20b65f14125e4218/cmake/extensions.cmake#L1316) checks to see if the specified option is valid for `C` and `CXX` and possibly `ASM`. But the `-Werror=implicit-int` option only produces a **warning**, not an **error** in the test, so it's allowed through (for C++).

I believe that the GCC release notes say that this will soon become a build-breaking error, not a warning, so it really does need to be fixed sometime.

But for the life of me I can't decipher that kind of CMakeFoo without some help...

vanwinkeljan, 2019-12-07:
@SebastianBoe could you provide some advice?

closed 2019-12-12 by galak

kyberdin, 2020-08-12:
I'm also encountering this issue when compiling a project with C++ sources, but with an additional warning:

```
cc1plus: warning: command line option '-Wno-pointer-sign' is valid for C/ObjC but not for C++
cc1plus: warning: '-Werror=' argument '-Werror=implicit-int' is not valid for C++
```

Not surprisingly, if I comment out [target_warnings.cmake:22](https://github.com/zephyrproject-rtos/zephyr/blob/master/cmake/compiler/gcc/target_warnings.cmake#L22) and [target_warnings.cmake:94](https://github.com/zephyrproject-rtos/zephyr/blob/master/cmake/compiler/gcc/target_warnings.cmake#L94) for GCC then the warnings no longer get printed.

Am I overlooking something in the configuration for disabling these warnings?

**OS:** `Ubuntu 18.04.5`
**Zephyr Version:** `2.3.0`
**Compiler Version:** `arm-zephyr-eabi-g++ (crosstool-NG 1.24.0.37-3f461da-dirty) 9.2.0`

PR #21242 (2019-12-12): cmake: toolchain: Don't add -Werror=implicit-int to CXX builds
We do compiler flag compatibility tests to be able to support many
different toolchains and flags in a scalable way. But the test is not
perfect and in these situations we we will need to hardcode whether a
flag is compatible or not.

To support this we have zephyr_compiler_check first check if the flag
is covered by a hardcoded test before doing an automatic test.

Currently the only hardcoded compatibilty is that -Werror=implicit-int
is not supported for CXX.

This fixes #21229
commit 01d8dc02 (2019-12-12): cmake: toolchain: Don't add -Werror=implicit-int to CXX builds

We do compiler flag compatibility tests to be able to support many
different toolchains and flags in a scalable way. But the test is not
perfect and in these situations we we will need to hardcode whether a
flag is compatible or not.

To support this we have zephyr_compiler_check first check if the flag
is covered by a hardcoded test before testing it.

Currently the only hardcoded compatibilty is that -Werror=implicit-int
is not supported for CXX.

This fixes #21229