zephyrproject-rtos/zephyr #21230: Remove unnecessary CMake target indirection

mbolivar, 2019-12-06:
**Is your enhancement proposal related to a problem? Please describe.**

As prep work for https://github.com/zephyrproject-rtos/zephyr/pull/13672, various changes were made to the build system which add extra layers of indirection for target names. One example commit is 1b86fb9da3.

Since this PR was rejected, the indirection is not necessary and will not be necessary, so it should be removed.

I'm not sure if any other indirection besides targets should be looked at.

**Describe the solution you'd like**

Remove extra indirection and refer to target names directly from CMake, e.g. `offsets_h` instead of `${OFFSETS_H_TARGET}`.

mbolivar, 2019-12-06:
FYI @carlescufi @SebastianBoe @tejlmand @hakonfam 

I'm not sure about the milestone, but this should be done "eventually".

SebastianBoe, 2019-12-09:
Agreed

tejlmand, 2019-12-13:
agreed/