zephyrproject-rtos/zephyr #21234: usb on sam0 does not enumerate after disable

sslupsky, 2019-12-07:
**Describe the bug**
Not sure if the API supports this but to save power I need to be able to disable the USB interface and then re-enable it.  In Arduino, I can call usb_detach() and then usb_attach() to do so.  When I attempt this using Zephyr usb_dc_detach() and usb_dc_attach(), the usb interface does not appear.  I attempted using usb_disable() and usb_enable() but have the same result. 

**To Reproduce**
Steps to reproduce the behavior:
1. build blinky and add usb_disable() and usb_enable()
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
I would like to be able to detach (when disconnected from the usb and running on battery) and attach again (when connected to usb).

**Impact**
Low power applications require that the USB be shut down.

sslupsky, 2019-12-07:
Some additional information I have on this issue.  

Using usb_enable() and usb_disable() appears to proceed further versus usb_dc_detach() and usb_dc_attach().  I believe the usb port is being enumerated now as I can see a log message indicating the usb_cdc_acm is configured (see below).  Further, I do see the device reappear as /dev/cu.usbmodem0_011.  However, when a host terminal connects the Zephyr shell running on the sam0 device does not respond.  I have used screen and CoolTerm as terminals on my host (macOS).

I have updated the issue title to reflect a better description of the problem.

Uncertain if this is a usb problem or a shell problem.

log output when usb_enable() called:
```
[00:03:14.047,149] <dbg> myApp.main: USB reconnected
[00:03:14.171,142] <dbg> usb_cdc_acm.cdc_acm_dev_status_cb: cfg 0x20005f40 status 1
[00:03:14.171,325] <dbg> usb_cdc_acm.cdc_acm_do_cb: USB device reset detected
[00:03:14.913,024] <dbg> usb_cdc_acm.cdc_acm_dev_status_cb: cfg 0x20005f40 status 3
[00:03:14.913,391] <dbg> usb_cdc_acm.cdc_acm_read_cb: ep 1 size 0 dev_data 0x20005754 rx_ringbuf space 1023
[00:03:14.913,574] <inf> usb_cdc_acm: USB device configured
```
log output when host terminal is reconnected to usb port:
```
[00:33:42.722,900] <dbg> usb_cdc_acm.cdc_acm_class_handle_req: CDC_SET_CONTROL_LINE_STATE 0x3
[00:33:42.724,395] <dbg> usb_cdc_acm.cdc_acm_class_handle_req: CDC_SET_LINE_CODING 9600 0 0 8
[00:33:42.726,135] <dbg> usb_cdc_acm.cdc_acm_class_handle_req: CDC_SET_LINE_CODING 9600 0 0 8
[00:33:42.727,844] <dbg> usb_cdc_acm.cdc_acm_class_handle_req: CDC_SET_LINE_CODING 9600 0 0 8
[00:33:42.729,705] <dbg> usb_cdc_acm.cdc_acm_class_handle_req: CDC_SET_LINE_CODING 9600 0 0 8
[00:33:42.731,018] <dbg> usb_cdc_acm.cdc_acm_class_handle_req: CDC_SET_LINE_CODING 9600 0 0 8
[00:33:42.732,818] <dbg> usb_cdc_acm.cdc_acm_class_handle_req: CDC_SET_LINE_CODING 9600 0 0 8
[00:33:42.734,130] <dbg> usb_cdc_acm.cdc_acm_class_handle_req: CDC_SET_LINE_CODING 9600 0 0 8
[00:33:42.735,931] <dbg> usb_cdc_acm.cdc_acm_class_handle_req: CDC_SET_LINE_CODING 9600 0 0 8
```
log output when a keypress on the host terminal occurs (note: four keypresses, 'size' is decrementing each time):
```
[00:33:58.173,980] <dbg> usb_cdc_acm.cdc_acm_read_cb: ep 1 size 1 dev_data 0x20005754 rx_ringbuf space 1023
[00:33:58.174,377] <dbg> usb_cdc_acm.cdc_acm_fifo_read: dev 0x20005c3c size 31 rx_ringbuf space 1022
[00:34:01.040,924] <dbg> usb_cdc_acm.cdc_acm_read_cb: ep 1 size 1 dev_data 0x20005754 rx_ringbuf space 1023
[00:34:01.041,320] <dbg> usb_cdc_acm.cdc_acm_fifo_read: dev 0x20005c3c size 30 rx_ringbuf space 1022
[00:34:01.681,060] <dbg> usb_cdc_acm.cdc_acm_read_cb: ep 1 size 1 dev_data 0x20005754 rx_ringbuf space 1023
[00:34:01.681,274] <dbg> usb_cdc_acm.cdc_acm_fifo_read: dev 0x20005c3c size 29 rx_ringbuf space 1022
[00:34:02.162,933] <dbg> usb_cdc_acm.cdc_acm_read_cb: ep 1 size 1 dev_data 0x20005754 rx_ringbuf space 1023
[00:34:02.163,330] <dbg> usb_cdc_acm.cdc_acm_fifo_read: dev 0x20005c3c size 28 rx_ringbuf space 1022
```

sslupsky, 2019-12-07:
This is the log output when the first keypress after usb_enable() on the host terminal occurs:
```
[00:01:32.164,337] <dbg> usb_cdc_acm.cdc_acm_read_cb: ep 1 size 1 dev_data 0x20005754 rx_ringbuf space 1023
[00:01:32.164,733] <dbg> usb_cdc_acm.cdc_acm_fifo_read: dev 0x20005c3c size 57 rx_ringbuf space 1022
[00:01:32.164,916] <dbg> usb_cdc_acm.cdc_acm_fifo_fill: dev_data 0x20005754 len 7 tx_ringbuf space 1023
[00:01:32.165,130] <dbg> usb_cdc_acm.tx_work_handler: Transfer is ongoing
```

sslupsky, 2019-12-07:
I modified the code around usb_disable() and usb_enable() to stop and start the shell but this made no difference.  

At this point, I am thinking this is a USB driver problem.

sslupsky, 2019-12-08:
This problem also occurs on a hardware cable detach and reattach without the usb_disable() and usb_enable().  That is to say, if I pull the usb cable and reattach, the usb port reappears on the host but there is no communication with the Zephyr shell on the device.

jfischer-phytec-iot, 2019-12-09:
usb_dc_disable() and usb_dc_enable() are from USB driver API, and may not used by the user. Support for `usb_enable` is somehow broken and will be fixed in #20375. Patches for usb_dc_sam0 are welcome.

sslupsky, 2019-12-09:
@jfischer-phytec-iot I read through that PR yesterday but I do not think this will resolve the issue here.  I think something is broken somewhere else.  I spent some time trying to determine if there was something about the shell or console that prevented those from reconnecting properly after the usb goes away.  I noticed several mutex sections in shell.c.  But that would not likely explain the lack of console then.   So, I keep coming back to usb.

I think I recall at one point that the end point address changed.  Normally, after boot the endpoint address is 0x82.  I see a lot of end point warnings to this address (#20987).  I think I recall that in one test where I disconnected the cable and reconnected it (my code did not call usb_enable() or usb_disable() directly), I saw a debug log message that referred to an endpoint with address 0x80.  However, I did not keep a record log message of that so I am not certain about the context of the observation.

Any suggestion of what to look for?  I am not very familiar with the inner workings of USB so I really do not know where to go with this.  This pretty much stop me in my tracks with Zephyr if I cannot enable() and disable() the usb port.  It nominally idles at about 6.5mA on a sam0.

Also, I got to thinking as I reflected about other issue and PR's related to device power management.  Shouldn't the USB driver of device power management built in?  Is this in the roadmap for these drivers?  Has anyone else implemented it?  Nordic maybe?

sslupsky, 2019-12-09:
@jfischer-phytec-iot Note in the log when the terminal reconnects after the usb is reattached, there are several CDC_SET_LINE_CODING 9600 0 0 8 in a row (I think there are 8 of them).  Does this suggest anything to you?  It looks to me like the host is trying to set the serial parameters and for some reason this happens 8 times suggesting that the host doesn't think the parameters were set?

sslupsky, 2019-12-09:
I turned on the debug log for usb_device and here is what the log looks like between the CDC_SET_LINE_CODING log output statements:
```
[00:12:26.775,604] <dbg> usb_cdc_acm.cdc_acm_class_handle_req: CDC_SET_LINE_CODING 9600 0 0 8
[00:12:26.775,787] <dbg> usb_device.usb_handle_control_transfer: >> usb_data_to_host(2)
[00:12:26.776,031] <dbg> usb_device.usb_handle_control_transfer: ep 80, status 2
[00:12:26.776,275] <dbg> usb_device.usb_handle_control_transfer: ep 0, status 0
[00:12:26.776,550] <dbg> usb_device.usb_handle_control_transfer: ep 0, status 1
[00:12:26.776,733] <dbg> usb_device.usb_handle_request: ** 1 **
[00:12:26.776,916] <dbg> usb_device.class_handler: bRequest 0x20, wIndex 0x0
```

jfischer-phytec-iot, 2019-12-09:
@sslupsky I have a SAMR21 Xplained board. I tested it with the latest master, I can disconnect the TARGET USB and connect it again (the board is still powered through EDBG USB), the board will be enumerated but it failed to set dtr/rts and after few attach/detach cycles it hard-faults. So we need to fix this issue first.

sslupsky, 2019-12-09:
@jfischer-phytec-iot I believe what you observed are the same symptoms I have encountered here.  The target appears to be enumerated and then fails to communicate.  

I have not tried multiple attach/detach cycles and experienced the hard fault yet.

sslupsky, 2019-12-09:
It appears that the line state (DTR) is being set after the usb is reattached:
```

### Terminal session was restarted on host

00> [00:01:04.016,326] <dbg> usb_device.usb_handle_control_transfer: ep 0, status 0
00> [00:01:04.016,693] <dbg> usb_device.usb_handle_request: ** 1 **
00> [00:01:04.016,876] <dbg> usb_device.class_ha
00> [00:01:04.016,876] <dbg> usb_device.class_handler: bRequest 0x22, wIndex 0x0
00> [00:01:04.017,059] <dbg> usb_cdc_acm.cdc_acm_class_handle_req: CDC_SET_CONTROL_LINE_STATE 0x3
00> [00:01:04.017,608] <dbg> usb_device.usb_handle_control_transfer: ep 80, status 2
00> [00:01:04.017,791] <dbg> usb_device.usb_handle_control_transfer: ep 0, status 0
00> [00:01:04.018,341] <dbg> usb_device.usb_handle_control_transfer: ep 0, status 1
00> [00:01:04.018,524] <dbg> usb_device.usb_handle_request: ** 1 **
00> [00:01:04.018,707] <dbg> usb_device.class_handler: bRequest 0x20, wIndex 0x0
00> [00:01:04.018,890] <dbg> usb_cdc_acm.cdc_acm_class_handle_req: CDC_SET_LINE_
00> [00:01:04.018,890] <dbg> usb_cdc_acm.cdc_acm_class_handle_req: CDC_SET_LINE_CODING 9600 0 0 8
00> [00:01:04.019,073] <dbg> usb_device.usb_handle_control_transfer: >> usb_data_to_host(2)
00> [00:01:04.019,256] <dbg> usb_device.usb_handle_control_transfer: ep 80, status 2
00> [00:01:04.019,653] <dbg> usb_device.usb_handle_control_transfer: ep 0, status 0
00> [00:01:04.019,836] <dbg> usb_device.usb_handle_control_transfer: ep 0, status 1
00> [00:01:04.020,019] <dbg> usb_device.usb_handle_request: ** 1 **
00> [00:01:04.020,202] <dbg> usb_device.class_handler: bRequest 0x20, wIndex 0x0
00> [00:01:04.020,385] <dbg> usb_cdc_acm.cdc_acm_class_handle_req: CDC_SET_LINE_CODING 9600 0 0 8
00> [00:01:04.020,568] <dbg> usb_device.usb_handle_control_transfer: >> usb_data_to_host(2)
00> [00:01:04.020,782] <dbg> usb_device.usb_handle_control_transfer: ep 80, status 2
00> [00:01:04.021,118] <dbg> usb_device.usb_handle_control_transfer: ep 0, status 0
00> [00:01:04.021,331] <dbg> usb_device.usb_handle_control_transfer: ep 0, status 1
00> [00:01:04.021,514] <dbg> usb_device.usb_handle_request: ** 1 **
```

sslupsky, 2019-12-09:
I noticed that after boot, the usb device performs some calls to "usb_descriptor":
```
00> [00:00:00.005,920] <dbg> usb_cdc_acm.cdc_acm_init: Device dev 0x2000683c dev_data 0x20006354 cfg 0x20006b40 added to devlist 0x200006c4
00> [00:00:00.006,103] <dbg> usb_descriptor.usb_get_device_descriptor: __usb_descriptor_start 0x20006a94
00> [00:00:00.006,286] <dbg> usb_descriptor.usb_get_device_descriptor: __usb_descriptor_end 0x20006b3d
00> [00:00:00.006,469] <dbg> usb_descriptor.usb_fix_descriptor: Configuration descriptor 0x20006aa6
00> [00:00:00.006,652] <dbg> usb
00> [00:00:00.006,652] <dbg> usb_descriptor.usb_fix_descriptor: Association descriptor 0x20006aaf
00> [00:00:00.006,835] <dbg> usb_descriptor.usb_fix_descriptor: Interface descriptor 0x20006ab7
00> [00:00:00.007,019] <dbg> usb_descriptor.usb_fix_descriptor: Endpoint descriptor 0x20006ad3
00> [00:00:00.007,202] <dbg> usb_descriptor.usb_validate_ep_cfg_data: Fixing EP address 80 -> 81
00> [00:00:00.007,385] <dbg> usb_descriptor.usb_validate_ep_cfg_data: endpoint 0x81
00> [00:00:00.007,568] <dbg> usb_descriptor.usb_fix_descriptor: Interface descriptor 0x20006ada
00> [00:00:00.007,751] <dbg> usb_descriptor.usb_fix_descriptor: Endpoint descriptor 0x20006ae3
00> [00:00:00.007,934] <dbg> usb_descriptor.usb_validate_ep_cfg_data: Fixing EP address 80 -> 82
00> [00:00:00.008,117] <dbg> usb_descriptor.usb_validate_ep_cfg_data: endpoint 0x82
00> [00:00:00.008,300] <dbg> usb_descriptor.usb_fix_descriptor: Endpoint descriptor 0x20006aea
00> [00:00:00.008,483] <dbg> usb_descriptor.usb_validate_ep_cfg_data: Fixing EP address 0 -> 1
00> [00:00:00.008,666] <dbg> usb_descriptor.usb_validate_ep_cfg_data: endpoint 0x1
00> [00:00:00.008,850] <dbg> usb_descriptor.usb_fix_descriptor: Now the wTotalLength is 75
00> [00:00:00.009,033] <dbg> usb_descriptor.ascii7_to_utf16le: idx_max 11, ascii_idx_max 5, buf 0x20006af7
00> [00:00:00.009,216] <dbg> usb_descriptor.ascii7_to_utf16le: char R : 52, idx 5 -> 11
```

Does this change the endpoint address?  the usb_descriptor calls do not happen again after the usb port is detached and then reattached.

When I look at the endpoint after boot and the end point after the reattach, they are different.  In the logs below, you can see after boot, it writes to EP 82.  After the reattach, it writes to EP 1. 

Here is the log after the initial boot:
```
### Return key was pressed in terminal

00> [00:00:21.604,431] <dbg> usb_cdc_acm.cdc_acm_write_cb: tx_ringbuf is empty
00> --- 48 messages dropped ---
00> [00:00:21.606,719] <dbg> usb_device.usb_transfer_work: transfer done, ep=82, status=0, size=1
00> [00:00:21.606,903] <dbg> usb_cdc_acm.cdc_acm_write_cb: ep 82: written 1 bytes dev_data 0x20006354
00> [00:00:21.607,086] <dbg> usb_cdc_acm.cdc_acm_write_cb: tx_ringbuf is empty
00> [00:00:21.607,299] <dbg> usb_cdc_acm.cdc_acm_fifo_fill: dev_data 0x20006354 len 7 tx_ringbuf space 1023
00> [00:00:21.607,513] <dbg> usb_cdc_acm.tx_work_handler: Got 7 bytes from ringbuffer send to ep 82
00> [00:00:21.607,696] <dbg> usb_device.usb_transfer: transfer start, ep=82, data=0x200063bc, dlen=7
00> [00:00:21.607,971] <dbg> usb_device.usb_transfer_work: transfer done, ep=82, status=0, size=7
00> [00:00:21.608,154] <dbg> usb_cdc_acm.cdc_acm_write_cb: ep 82: written 7 bytes dev_data 0x20006354
00> [00:00:21.608,337] <dbg> usb_cdc_acm.cdc_acm_write_cb: tx_ringbuf is empty
00> [00:00:21.608,612] <dbg> usb_cdc_acm.cdc_acm_fifo_fill: dev_data 0x20006354 len 1 tx_ringbuf space 1023
00> --- 176 messages dropped ---
00> [00:00:21.664,031] <dbg> usb_cdc_acm.cdc_acm_fifo_fill: dev_data 0x20006354 len 7 tx_ringbuf space 1023
```

and there is the same thing after the reattach:
```
### Return key was pressed on host terminal

00> [00:01:13.604,003] <dbg> usb_device.usb_transfer_work: transfer done, ep=01, status=0, size=1
00> [00:01:13.604,370] <dbg> usb_cdc_acm.cdc_acm_read_cb: ep 1 size 1 dev_data 0x20006354 rx_ringbuf space 1023
00> [00:01:13.604,553] <dbg> usb_device.usb_transfer: transfer start, ep=01, data=0x2000637c, dlen=64
00> [00:01:13.604,766] <dbg> usb_cdc_acm.cdc_acm_fifo_read: dev 0x2000683c size 57 rx_ringbuf space 1022
00> [00:01:13.604,949] <dbg> usb_cdc_acm.cdc_acm_fifo_fill: dev_data 0x20006354 len 7 tx_ringbuf space 1023
00> [00:01:13.605,163] <dbg> usb_cdc_acm.tx_work_handler: Transfer is ongoing
00> [00:01:18.556,121] <dbg> myApp.led: LED: Blink

```

sslupsky, 2019-12-10:
@jfischer-phytec-iot I think I have figured this out.  It does not appear to be a sam0 specific issue.  I suppose sam0 could exacerbate the issue some way but it is not obvious to me that it does.  I think this issue likely occurs on other SOC's as well.

The issue is related to the disconnect notification.  I observed that I was not receiving a USB_DC_DISCONNECT notice.  Upon a deeper review and in consideration of the "Transfer is ongoing" message, I determined there is an issue where a usb_transfer() can get stuck if the transfer is flagged busy.  Pulling the plug so to speak on the USB can cut off a transfer mid stream and leave this flag in its busy state.  Moreover, when a USB_DC_RESET is triggered, the transfers are not reset.

Here is where the issue manifests in usb_device.c:
```
static void forward_status_cb(enum usb_dc_status_code status, const u8_t *param)
{
	size_t size = (__usb_data_end - __usb_data_start);

	if (status == USB_DC_DISCONNECTED || status == USB_DC_SUSPEND  ) {
		if (usb_dev.configured) {
			usb_cancel_transfers();
			if (status == USB_DC_DISCONNECTED) {
				foreach_ep(disable_interface_ep);
				usb_dev.configured = false;
			}
		}
	}
```
If there is a "USB_DC_DISCONNECTED" or "USB_DC_SUSPEND" status, the transfers are cancelled.  However, for "USB_DC_RESET", they are not cancelled.  I added ```|| status == USB_DC_RESET``` to the first conditional and the CDC_ACM started to receive the transfers.  I am uncertain if I also need to add this check to the second conditional.  Could you please comment if you think it is required here as well.

sslupsky, 2019-12-10:
The patch above addresses the situation where you physically disconnect the cable and reconnect.  

The port does not enumerate when you call usb_disable() and then usb_enable().  This is the original issue and requires further investigation.

sslupsky, 2019-12-10:
I did not notice earlier but apparently, this patch only works when the debugger to connected to the debug port and a debug session is running.  Not sure why that makes a difference.  If I shut down the debug session and/or disconnect the debugger, then the USB port does not enumerate when reconnected.

carlescufi, 2019-12-11:
@benpicco would you mind taking a look at this issue if you have some time? @jfischer-phytec-iot has reproduced it locally.

benpicco, 2019-12-15:
@carlescufi I'm afraid I'm not familiar enough with the USB stack and a bit short in time right now.

closed 2020-08-28 by github-actions[bot]

KubaFYI, 2020-09-24:
@sslupsky are you planning to make a PR with the patch in [this comment of yours](https://github.com/zephyrproject-rtos/zephyr/issues/21234#issuecomment-564040983)?

sslupsky, 2022-08-18:
@KubaFYI See PR #49190 which addresses the original issue of attach and detach.