zephyrproject-rtos/zephyr #21238: Improve Zephyr HCI VS extension detection

joerchan, 2019-12-09:
The heuristic should be that if the Version is 5.0 or later, and the controller does not have a valid public address. It's not a very good heuristic in my opinion. It works for this case because we only use to check for random static identity address. But in the case the controller supports many extensions to HCI but still has a valid Bluetooth address we would not be able to use these features.

jhedberg
Yep, there really isn't any good solution here (if there was we wouldn't need a Kconfig to enable the heuristics). One thing we do in Linux is to check the reported manufacturer ID, however Zephyr allows overriding it so it's not particularly helpful.

jhedberg 
Hmm, maybe we should require the LF manufacturer ID by default for the heuristics? If someone wants to pair up Zephyr host with a controller with non-LF manufacturer ID then they won't be able to use the heuristics but just have the host assume that it's talking to a Zephyr?

 @jhedberg
IIRC the Zephyr controller was introduced before LF had its SIG-assigned ID, which is why this code isn't taking advantage of it. Perhaps we can simply assume no-one is going to pair up a recent Zephyr host with an ancient Zephyr controller.

joerchan 
I think the LF company ID is too restrictive. It would prevent us from having other controllers implement the Zephyr HCI specified VS extension. In my opinion the best approach would be to have a kind of magic number. But that would mean adding a read_magic command/event.

jhedberg, 2019-12-09:
> I think the LF company ID is too restrictive. It would prevent us from having other controllers implement the Zephyr HCI specified VS extension.

I'm not sure I agree. The heuristic is not something I'd expect to be enabled in most cases. Most cases you're creating a product and you know exactly what host you pair up with what controller, so the heuristic is just unnecessary code. The biggest use case I see is Qemu/native_posix where you use a controller plugged into the host and don't want your host blowing up because of a non-Zephyr controller. In that case you can either disable the heuristic (if you know you have a Zephyr controller) or make sure the Zephyr build you have on the controller uses the LF ID.

I'm sure there are aspects and use cases I haven't thought of, but this is anyway my current understanding.

joerchan, 2020-12-08:
@jhedberg The Company ID approach with LF would not work for our out-of-tree Bluetooth LE Controller which implements the Zephyr HCI VS extensions.

There hasn't been any issues with the VS HCI system since I reported this issue, so I'm considering to just close this. Any objections?

jhedberg, 2020-12-08:
No objections to closing this

closed 2020-12-09 by joerchan