zephyrproject-rtos/zephyr #21239: devicetree: Generation of the child-bindigs items as a common static initializer

KAGA164, 2019-12-09:
Currently I am thinking about generic gpio led driver, but It is hard to generate led data for it without led description duplication in DTS. I found the child-bindings construction, it is used among others for leds configurations generation.

Now It looks like:

```
	leds {
		compatible = "gpio-leds";
		led0: led_0 {
			gpios = <&gpio0 17 GPIO_INT_ACTIVE_LOW>;
			label = "Green LED 0";
		};
		led1: led_1 {
			gpios = <&gpio0 18 GPIO_INT_ACTIVE_LOW>;
			label = "Green LED 1";
		};
		led2: led_2 {
			gpios = <&gpio0 19 GPIO_INT_ACTIVE_LOW>;
			label = "Green LED 2";
		};
		led3: led_3 {
			gpios = <&gpio0 20 GPIO_INT_ACTIVE_LOW>;
			label = "Green LED 3";
		};
	};
```

and this is generate for each child node like that:
```

#define DT_GPIO_LEDS_LED_0_GPIOS_CONTROLLER         "GPIO_0"
#define DT_ALIAS_LED0_GPIOS_CONTROLLER              DT_GPIO_LEDS_LED_0_GPIOS_CONTROLLER
#define DT_GPIO_LEDS_LED0_GPIOS_CONTROLLER          DT_GPIO_LEDS_LED_0_GPIOS_CONTROLLER
#define DT_GPIO_LEDS_LED_0_GPIOS_PIN                13
#define DT_ALIAS_LED0_GPIOS_PIN                     DT_GPIO_LEDS_LED_0_GPIOS_PIN
#define DT_GPIO_LEDS_LED0_GPIOS_PIN                 DT_GPIO_LEDS_LED_0_GPIOS_PIN
#define DT_GPIO_LEDS_LED_0_GPIOS_FLAGS              0
#define DT_ALIAS_LED0_GPIOS_FLAGS                   DT_GPIO_LEDS_LED_0_GPIOS_FLAGS
#define DT_GPIO_LEDS_LED0_GPIOS_FLAGS               DT_GPIO_LEDS_LED_0_GPIOS_FLAGS
#define DT_GPIO_LEDS_LED_0_GPIOS                    {"GPIO_0", 13, 0}
#define DT_ALIAS_LED0_GPIOS                         DT_GPIO_LEDS_LED_0_GPIOS
#define DT_GPIO_LEDS_LED0_GPIOS                     DT_GPIO_LEDS_LED_0_GPIOS
/* Human readable string describing the device (used by Zephyr for API name) */
#define DT_GPIO_LEDS_LED_0_LABEL                    "Green LED 0"
#define DT_ALIAS_LED0_LABEL                         DT_GPIO_LEDS_LED_0_LABEL
#define DT_GPIO_LEDS_LED0_LABEL                     DT_GPIO_LEDS_LED_0_LABEL
```

The purpose of my proposition is to add a initializer to every node which contatins all parameters from DTS, for example:

#define DT_GPIO_LEDS_LED0 {DT_GPIO_LEDS_LED_0_GPIOS , DT_GPIO_LEDS_LED_0_LABEL , other_param ....}

And in parent node add:

#define  DT_GPIO_LEDS {DT_GPIO_LEDS_LED0, DT_GPIO_LEDS_LED1, DT_GPIO_LEDS_LEDx}

Then we can do this:

`static stuct gpio_leds leds[] = DT_GPIO_LEDS;`

This should be very useful in context of development the new generic drivers, because we have all needed data to initalize it at the compilation stage.

KAGA164, 2019-12-09:
@pabigot

ulfalizer, 2019-12-10:
It's nice if `child-binding` itself can just be a generic mechanism for assigning bindings to child nodes, with no special logic applied (i.e., have it work just as-if you had assigned the binding manually to all the child nodes via `compatible`).

Could always special-case e.g. LED nodes though, if that'd be helpful.

KAGA164, 2019-12-10:
I would like to have it for LEDs but I do not see any reason to not provide this feature to all child-bindings objects. It is enough for me to have this define generations for LED at this moment. This do not break this generic mechanis for binding, it may add only few useful initializers.

pabigot, 2019-12-10:
See also #21243 which proposes `struct gpio_phandle_specifier` as a defined struct known to be compatible with the initializer lists produced by the devicetree scripting.

ulfalizer, 2019-12-10:
@KAGA164 
Suspect it can be handled in some way besides special-casing `child-binding`. Nice if it can stay as a dead simple "gives a binding to child nodes", with no extra magic beyond that on its own.

ulfalizer, 2019-12-12:
Looking into this now.

ulfalizer, 2019-12-14:
@KAGA164 
Finished implementing it: https://github.com/zephyrproject-rtos/zephyr/pull/21393

Tell me if it works the way you expect and if you run into any issues.

gmarull, 2022-11-21:
We already have a way to solve this:

https://github.com/zephyrproject-rtos/zephyr/blob/144d30eb47b97090894c2b439712c124320f8eb4/drivers/led/led_gpio.c#L88-L90

closed 2022-11-21 by gmarull