zephyrproject-rtos/zephyr #21240: Error west flash

pankaj18, 2019-12-09:
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
/Users/pankajjadhav/zephyrproject/zephyr/boards/arm/stm32_min_dev/support/openocd.cfg:1: Error: Can't find interface/stlink.cfg
in procedure 'script' 
at file "embedded:startup.tcl", line 60
at file "/Users/pankajjadhav/zephyrproject/zephyr/boards/arm/stm32_min_dev/support/openocd.cfg", line 1
ERROR: command exited with status 1: /usr/local/bin/openocd -f /Users/pankajjadhav/zephyrproject/zephyr/boards/arm/stm32_min_dev/support/openocd.cfg -c init -c targets -c 'reset halt' -c 'flash write_image erase /Users/pankajjadhav/zephyrproject/zephyr/build/zephyr/zephyr.hex' -c 'reset halt' -c 'verify_image /Users/pankajjadhav/zephyrproject/zephyr/build/zephyr/zephyr.hex' -c 'reset run' -c shutdown

aescolar, 2019-12-09:
cc @mbolivar

chrta, 2019-12-09:
Was an stlink programmer properly connected?

I believe there is no on-board programmer, but this is not clearly documented in the board documentation of zephyr.

mbolivar, 2019-12-09:
> I believe there is no on-board programmer, but this is not clearly documented in the board documentation of zephyr.

Looks like you're right: https://embedjournal.com/iot-rtos-zephyr-stm32-minimum-system-development-board/#hardware-setup

@pankaj18 can you please answer @chrta 's question?

> Was an stlink programmer properly connected?

Please also read the embedjournal link above if you need more information.

closed 2019-12-10 by pankaj18

yhyuan, 2020-07-17:
I am having the same issue. It seems openocd fails to find the stlink.cfg. I am not sure how to fix it.