zephyrproject-rtos/zephyr #21252: Getting Started Guide: west update procedure could lead to a not a good starting point

dstroe2000, 2019-12-09:
Following the guidance of the **Getting Started Guide** ([https://docs.zephyrproject.org/latest/getting_started/index.html](url)), section **Get the source code**, could lead to download an unstable tip of the repository.
It should be possible to select rather a stable release, as this might be a better starting point than the always latest version that could be not yet fully proofed.

dbkinder, 2019-12-09:
I suppose we could document that the GSG's setup using ``west init zephyrproject`` will pull the current master branch of the project.  If you want to work with a specific tagged release, you would instead specify ``west init --mr v1.14.1 zephyrproject``, for example. 

What does the developer community recommend:  
- pulling the latest LTS release (v1.14.1), 
- the latest tagged release (v2.1), 
- or master (default)?  I opted for master since it won't need updating as tagged releases are made)

The developer can always change to another tagged version of Zephyr via ``git checkout`` and a ``west update``, but that might be an "advanced topic".

jmgao, 2019-12-10:
> I opted for master since it won't need updating as tagged releases are made

An intermediate option would be to maintain a `stable` branch (or multiple, for latest release, latest LTS release, etc.) that gets the latest tagged release merged into it regularly.

dleach02, 2019-12-11:
I think the request here is to add to the documentation information on how to initialize the build environment to a specific version. Each release point is supposed to be a stable point in time (until the next release) for folks to start from.

dueringa, 2021-02-19:
The documentation change seems relatively trivial to me (correct me if I'm wrong please), I'd be willing to do a PR. As @dbkinder mentioned - should the instruction just mention the latest release (currently zephyr-v2.5.0), bearing in mind the tag in the doc should be updated on each release?