zephyrproject-rtos/zephyr #21253: 2.2 Release Checklist

dleach02, 2019-12-09:
Add items as required
--------
- [ ] Major enhancements:
   - [ ]  Make newlib libc the default c library: #3102 (Pushed to 2.1.0) 

- [ ] Pre RC1 Steps
   - [ ] Verify that all external components and external dependencies are up to date, for example 
      - mbedTLS
      - tinycrypt
      - FatFS
   - [ ] Check known vulnerabilities, and fix exploitable vulnerabilities or verify them as un-exploitable (Security working group).

See https://github.com/zephyrproject-rtos/zephyr/wiki/Release-procedure for tagging details

--------
- [ ] Release Notes WIP
  - [ ] Create draft doc/release-notes-2.2.rst with feature summary from https://www.zephyrproject.org/developers/#releases-overview
  - [ ] Update doc/release-notes-2.2.rst detailed sections (relative to 2.1.0 release) as important issues are addressed (subsystem owners)
--------
- [ ] Release data
  - [ ] Update version in VERSION
  - [ ] Run sanitycheck with --all --release and checking resulting file scripts/sanity_chk/sanity_last_release.csv
--------
- [ ] Finalize Release Notes
  - [ ] Add list of GitHub issues (spell check issue list and fix in GitHub as needed)
  - [ ] Review doc/releases/release-notes-2.2.rst overview summary and details sections (see Sign-off below)
    - [ ] Request summary details from code owners
  - [ ] ~~Update doc/release-notes/index.rst (add reference to 2.2 notes)~~
  - [ ] ~~Update doc/index.rst (add reference to doc/2.2.0)~~
  - [ ] Update doc/conf.py (add version to pick list)
  - [ ] Update doc/LICENSING.rst for new components that do not have a license
  - [ ] Create release and add notes on https://github.com/zephyrproject-rtos/zephyr/releases
  - [ ] Final tagged release as zephyr-v2.1.0 (with release notes)
  - [ ] Update https://www.zephyrproject.org/developers/#releases-overview
  - [ ] Update https://www.zephyrproject.org/developers/#downloads (and git checkout version example)
  - [ ] Update doc generation (in future should be automatic based on tag) ping @nashif
  - [ ] Email devel@lists.zephyrproject.org & announce@lists.zephyrproject.org lists about release
--------
- [ ]  Final Release Notes Sign-off
  - Subsystems
    - [ ] Networking - @jukkar 
    - [ ] Bluetooth - @jhedberg , @carlescufi 
  - Architectures
    - [ ] ARC - @ruuddw 
    - [ ] ARM - @galak, @MaureenHelm 
    - [ ] X86 - @nashif, @andrewboie 
    - [ ] RISCV - @pgielda @nategraff-sifive
    - [ ] Xtensa, NIOS-II - @nashif, @andrewboie 
  - [ ] Security - @d3zd3z 
--------
- [ ] After Release
  - [ ] Update patchlevel in Makefile (PATCHLEVEL = 99)
  - [ ] Marketing Blog for announcing release (https://www.zephyrproject.org/blog)
  - [ ] Create branch for 2.2.x releases, update github protections for branch
  - [ ] Create checklist issue for next release
  - [ ] Add a comment to this issue linking to the next release checklist issue
  - [ ] Update https://www.zephyrproject.org with latest stats from Github [@dbkinder]
  - [ ] Move all PRs/Issues in the 2.2 milestone to the next milestone
  - [ ] Close v2.2.0 milestone
  - [ ] Create v2.2.1 milestone

dleach02, 2020-03-04:
The main website changed from (or during) the last release so I'm not sure https://www.zephyrproject.org/developers/#downloads has a new place to resolve to. Probably should remove it from the checklist unless someone knows where this should resolve to.

jhedberg, 2020-03-10:
Next release's checklist is here: https://github.com/zephyrproject-rtos/zephyr/issues/22637

jhedberg, 2020-03-13:
Closing since all items have been completed (blog post text was just sent to the marketing team)

closed 2020-03-13 by jhedberg