zephyrproject-rtos/zephyr #21278: How to use pwm in nrf52832 for rgb led

pankaj18, 2019-12-10:
The sample code is given for Hexiwear K64. How to use this code for nrf52832 board and external  RGB led using pwm. Which pins are used and how to change given code

#if defined(DT_ALIAS_RED_PWM_LED_PWMS_CONTROLLER) && \
      defined(DT_ALIAS_RED_PWM_LED_PWMS_CHANNEL) && \
      defined(DT_ALIAS_GREEN_PWM_LED_PWMS_CONTROLLER) && \
      defined(DT_ALIAS_GREEN_PWM_LED_PWMS_CHANNEL) && \
      defined(DT_ALIAS_BLUE_PWM_LED_PWMS_CONTROLLER) && \
      defined(DT_ALIAS_BLUE_PWM_LED_PWMS_CHANNEL)
/* Get the defines from dt (based on aliases 'red-pwm-led', 'green-pwm-led' &
 * 'blue-pwm-led')
 */
#define PWM_DEV0	DT_ALIAS_RED_PWM_LED_PWMS_CONTROLLER
#define PWM_CH0		DT_ALIAS_RED_PWM_LED_PWMS_CHANNEL
#define PWM_DEV1	DT_ALIAS_GREEN_PWM_LED_PWMS_CONTROLLER
#define PWM_CH1		DT_ALIAS_GREEN_PWM_LED_PWMS_CHANNEL
#define PWM_DEV2	DT_ALIAS_BLUE_PWM_LED_PWMS_CONTROLLER
#define PWM_CH2		DT_ALIAS_BLUE_PWM_LED_PWMS_CHANNEL
#else
#error "Choose supported board or add new board for the application"
#endif

carlescufi, 2019-12-11:
@mbolivar perhaps you can shed some light here?

mbolivar, 2019-12-11:
@pankaj18 where is this code from? I'm really not clear what the question is.

henrikbrixandersen, 2019-12-12:
The code in question (rgb_led sample, likely) depends on device tree aliases for the pwm-leds nodes corresponding to the RGB LED to work.

Which board are you trying to compile for?

mbolivar, 2019-12-12:
> (rgb_led sample, likely)

Yep, that makes sense.

@pankaj18 did you read the device tree documentation?

https://docs.zephyrproject.org/latest/guides/dts/index.html

It explains what an "alias" is.

What you need to set an alias to depends on your board.

mbolivar, 2020-01-08:
> @pankaj18 did you read the device tree documentation?

Ping

henrikbrixandersen, 2020-02-06:
@pankaj18 Thank you for submitting the question. If the above answers did not solve your problem, please reopen this issue with any follow-up questions.

closed 2020-02-06 by henrikbrixandersen