zephyrproject-rtos/zephyr #21293: adding timeout the I2C read/write functions for the stm32 port

StefJar, 2019-12-11:
The current implementation of the stm32 I2C driver assumes that the I2C device is fully operational for any IO operation.
If in  whatever case the IC gets broken or maybe not soldered properly while production the current implementation runs into blocking mode. Especially if the I2C read/writes are at the startup this leads to a hangup of the whole device.
 
IMHO it would be great, if we can setup timeout for the IO operations.

I put a link to the code that is blocking in case of error:

https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/i2c/i2c_ll_stm32_v1.c#L567

erwango, 2019-12-11:
Seems legitimate.
Behavior is specific to `I2C_STM32_INTERRUPT` config, as we have timeout when interrupts are not used. It would make sense to trigger an error here as well and have similar behavior in 2 configurations.

ydamigos, 2019-12-11:
@erwango I opened PR  #21308 to address the issue. I did some quick tests but it needs more testing.

avisconti, 2019-12-12:
Well, it seems somehow related to #20639, which is a bug that I opened recently, but for which we never understood the root cause. Actually what I have observed at that time was in fact that the I2C driver was stuck on a k_sem_take() because for some reason a k_sem_give() was never executed.
But the issue was not easily reproducible.

I may review and re-test with #21308.

erwango, 2021-06-04:
Fixed by https://github.com/zephyrproject-rtos/zephyr/pull/32777 and #27291

closed 2021-06-04 by erwango