zephyrproject-rtos/zephyr #21299: bluetooth: Controller does not release buffer on central side after peripheral reset

KAGA164, 2019-12-11:
I probably found a problem with connection between two Bluetooth device after I three time reset the peripheral side. This is always the third reset which causes problem with connection.  In this file:
https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/bluetooth/controller/ll_sw/ull_conn.c

is 

```
static struct {                                                                                                                                                        
            void *free;                                                                                                                                                    
            u8_t pool[CONN_TX_CTRL_BUF_SIZE * CONN_TX_CTRL_BUFFERS];                                                                                                       
} mem_conn_tx_ctrl;
```

where CONN_TX_CTRL_BUFFERS is defined:
```

/**                                                                                                                                                                    
 * One connection may take up to 4 TX buffers for procedures                                                                                                           
 * simultaneously, for example 2 for encryption, 1 for termination,                                                                                                    
 * and 1 one that is in flight and has not been returned to the pool                                                                                                   
 */                                                                                                                                                                    
#define CONN_TX_CTRL_BUFFERS  (4 * CONFIG_BT_CTLR_LLCP_CONN) 
```     

In default configuration this is equal to 4

So I discovered that every accidental peripheral device disconnection like reset or power down,
takes one buffer from this pool and this buffer is probably not release any more. So when I reset peripheral side third time, on reconnection at central side I am beyond this buffer when central device want to send link layer stuff for example phy update and connection is terminated. After that device try to connect and immediately disconnect in loop util one of them get the Hard Fault error.

I try to increase the CONN_TX_CTRL_BUFFERS  to 10 then I can reset the peripheral device nine time.

I use two nrf52_pca10040 board. One is the peripheral side. I use secutity level 2 with encryption and bonding.

Steps to reproduce the behavior:
1. Connect
1. pair and bond with security level 2
2. Wait until central side starts receiving notifications
3. Reset the peripheral board
4. Repeat whole procedure after third reset, boards are not able to have a stable connection because
peripheral board is disconnected(reason 34) on central (reason 8) (but it seems like central send corrupted link layer frame) in loop

KAGA164, 2019-12-11:
@joerchan

joerchan, 2019-12-11:
Thanks @KAGA164.

@cvinayak Is this perhaps the same as NCSDK-3751?

KAGA164, 2019-12-11:
@joerchan It looks like the same test case. I stuck mostly on PHY Update which used the same buffer like that request in NCSDK-3751 but the rest of issue descriptions is the same

cvinayak, 2019-12-12:
@KAGA164 Could you please provide me the samples and the commit hash to reproduce the issue? I will check the samples/bluetooth/peripheral vs samples/bluetooth/central_hr with added bt_conn_security_set, hope I can reproduce. But it is always best if the issue report has better details for reproducing, to speed up on the resolution.

PR #21374 (2019-12-16): Bluetooth: controller: split: Fix Tx Ctrl PDU leak
Overlapping Feature Exchange requested by host with
Encryption Setup requested by the application caused the
controller to corrupt its Tx queue leading to Tx Ctrl PDU
buffers from leaking from the system.

Fixes #21299.
commit f7c890e0 (2019-12-16): Bluetooth: controller: split: Fix Tx Ctrl PDU leak

Overlapping Feature Exchange requested by host with
Encryption Setup requested by the application caused the
controller to corrupt its Tx queue leading to Tx Ctrl PDU
buffers from leaking from the system.

Fixes #21299.
closed 2019-12-16 by aescolar