zephyrproject-rtos/zephyr #21300: west (pyocd) flash failing on bbc_microbit

pabigot, 2019-12-11:
I mentioned this on #general a while back and it worked for other people, so it may be specific to my environment, which is Pop!_OS (Ubuntu) 19.04.

Attempts to flash images with `west flash` fail for me with `bbc_microbit`.  Copying the hex file to the mounted directory works.  I suspect a problem with pyocd or installed USB libraries but haven't been able to nail it down.

```
tirzah[122]$ west -v flash
ZEPHYR_BASE=/mnt/nordic/zp/zephyr (origin: env)
-- west flash: rebuilding
ninja: no work to do.
-- west flash: using runner pyocd
-- runners.pyocd: Flashing file: /mnt/nordic/zp/zephyr/samples/hello_world/build/zephyr/zephyr.hex
runners.pyocd: pyocd flash -e sector -t nrf51 /mnt/nordic/zp/zephyr/samples/hello_world/build/zephyr/zephyr.hex
0001338:CRITICAL:__main__:uncaught exception: [Errno 110] Operation timed out
Traceback (most recent call last):
  File "/home/pab/.local/lib/python3.7/site-packages/pyocd/__main__.py", line 343, in run
    self._COMMANDS[self._args.cmd](self)
  File "/home/pab/.local/lib/python3.7/site-packages/pyocd/__main__.py", line 473, in do_flash
    with session:
  File "/home/pab/.local/lib/python3.7/site-packages/pyocd/core/session.py", line 287, in __enter__
    self.open()
  File "/home/pab/.local/lib/python3.7/site-packages/pyocd/core/session.py", line 363, in open
    self._probe.open()
  File "/home/pab/.local/lib/python3.7/site-packages/pyocd/probe/cmsis_dap_probe.py", line 141, in open
    self._link.open()
  File "/home/pab/.local/lib/python3.7/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 580, in open
    self._packet_count = self._protocol.dap_info(self.ID.MAX_PACKET_COUNT)
  File "/home/pab/.local/lib/python3.7/site-packages/pyocd/probe/pydapaccess/cmsis_dap_core.py", line 132, in dap_info
    self.interface.write(cmd)
  File "/home/pab/.local/lib/python3.7/site-packages/pyocd/probe/pydapaccess/interface/pyusb_backend.py", line 193, in write
    self.ep_out.write(data)
  File "/home/pab/.local/lib/python3.7/site-packages/usb/core.py", line 387, in write
    return self.device.write(self, data, timeout)
  File "/home/pab/.local/lib/python3.7/site-packages/usb/core.py", line 948, in write
    self.__get_timeout(timeout)
  File "/home/pab/.local/lib/python3.7/site-packages/usb/backend/libusb1.py", line 842, in intr_write
    timeout)
  File "/home/pab/.local/lib/python3.7/site-packages/usb/backend/libusb1.py", line 920, in __write
    _check(retval)
  File "/home/pab/.local/lib/python3.7/site-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 110] Operation timed out
ERROR: command exited with status 1: pyocd flash -e sector -t nrf51 /mnt/nordic/zp/zephyr/samples/hello_world/build/zephyr/zephyr.hex
Traceback (most recent call last):
  File "/home/pab/.local/lib/python3.7/site-packages/west/main.py", line 583, in main
    args.handler(args, unknown)
  File "/home/pab/.local/lib/python3.7/site-packages/west/main.py", line 334, in ext_command_handler
    command.run(args, unknown, topdir, manifest=manifest)
  File "/home/pab/.local/lib/python3.7/site-packages/west/commands/__init__.py", line 118, in run
    self.do_run(args, unknown)
  File "/mnt/nordic/zp/zephyr/scripts/west_commands/flash.py", line 32, in do_run
    'ZEPHYR_BOARD_FLASH_RUNNER')
  File "/mnt/nordic/zp/zephyr/scripts/west_commands/run_common.py", line 290, in do_run_common
    runner.run(command_name)
  File "/mnt/nordic/zp/zephyr/scripts/west_commands/runners/core.py", line 441, in run
    self.do_run(command, **kwargs)
  File "/mnt/nordic/zp/zephyr/scripts/west_commands/runners/pyocd.py", line 116, in do_run
    self.flash(**kwargs)
  File "/mnt/nordic/zp/zephyr/scripts/west_commands/runners/pyocd.py", line 146, in flash
    self.check_call(cmd)
  File "/mnt/nordic/zp/zephyr/scripts/west_commands/runners/core.py", line 513, in check_call
    subprocess.check_call(cmd)
  File "/usr/lib/python3.7/subprocess.py", line 347, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['pyocd', 'flash', '-e', 'sector', '-t', 'nrf51', '/mnt/nordic/zp/zephyr/samples/hello_world/build/zephyr/zephyr.hex']' returned non-zero exit status 1.
tirzah[123]$ cp /mnt/nordic/zp/zephyr/samples/hello_world/build/zephyr/zephyr.hex /media/pab/MICROBIT/
tirzah[124]$ uname -a
Linux tirzah 5.3.0-22-generic #24+system76~1573659475~19.04~26b2022-Ubuntu SMP Wed Nov 13 20:0 x86_64 x86_64 x86_64 GNU/Linux
tirzah[125]$ cat /etc/lsb-release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=19.04
DISTRIB_CODENAME=disco
DISTRIB_DESCRIPTION="Pop!_OS 19.04"
tirzah[126]$ dpkg -l | grep usb
ii  ippusbxd                                                    1.33-1                                                           amd64        Daemon for IPP USB printer support
ii  libgusb2:amd64                                              0.3.0-1                                                          amd64        GLib wrapper around libusb1
ii  libhidapi-libusb0:amd64                                     0.8.0~rc1+git20140818.d17db57+dfsg-2                             amd64        Multi-Platform library for communication with HID devices (libusb backend)
ii  libusb-0.1-4:amd64                                          2:0.1.12-32                                                      amd64        userspace USB programming library
ii  libusb-1.0-0:amd64                                          2:1.0.22-2                                                       amd64        userspace USB programming library
ii  libusb-1.0-0-dev:amd64                                      2:1.0.22-2                                                       amd64        userspace USB programming library development files
ii  libusb-1.0-doc                                              2:1.0.22-2                                                       all          documentation for userspace USB programming
ii  libusbmuxd4:amd64                                           1.1.0~git20181007.07a493a-1                                      amd64        USB multiplexor daemon for iPhone and iPod Touch devices - library
ii  libusbredirhost1:amd64                                      0.8.0-1                                                          amd64        Implementing the usb-host (*) side of a usbredir connection (runtime)
ii  libusbredirparser1:amd64                                    0.8.0-1                                                          amd64        Parser for the usbredir protocol (runtime)
ii  spice-client-glib-usb-acl-helper                            0.35-2                                                           amd64        Helper tool to validate usb ACLs
ii  usb-modeswitch                                              2.5.2+repack0-2ubuntu1                                           amd64        mode switching tool for controlling "flip flop" USB devices
ii  usb-modeswitch-data                                         20170806-2                                                       all          mode switching data for usb-modeswitch
ii  usb.ids                                                     2019.02.23-1                                                     all          USB ID Repository
ii  usbmuxd                                                     1.1.1~git20181007.f838cf6-1                                      amd64        USB multiplexor daemon for iPhone and iPod Touch devices
ii  usbutils                                                    1:010-3                                                          amd64        Linux USB utilities
tirzah[127]$ 
```

carlescufi, 2019-12-11:
@pabigot I tested this locally and it works fine. Can you try [updating the firmware](https://microbit.org/guide/firmware/)?

carlescufi, 2019-12-11:
@pabigot also, to rule out any other issues, can you also try `west flash --flash-opt='-e=chip'`

carlescufi, 2019-12-11:
FWIW here are my versions of the packages that seem relevant:

```
libusb1                        1.7.1 
pyocd                          0.21.0  
```

pabigot, 2019-12-11:
Firmware is already 253, `--flash-opt='-e=chip'` had no effect, pyocd 0.21, 0.22, and 0.23 (default) all behave the same.  This did work on my previous development machine; not sure if it broke with the newer one, or due to updates to the newer one.

mbolivar, 2019-12-11:
It looks like the pyocd subprocess itself is failing due to a USB timeout.

```
pyocd flash -e sector -t nrf51 /mnt/nordic/zp/zephyr/samples/hello_world/build/zephyr/zephyr.hex
```

There's nothing wrong with that command line that I can see, so I think this is not a west bug.

The same command works fine for me with pyocd 0.23 installed in a virtualenv, while you got:

```
0001338:CRITICAL:__main__:uncaught exception: [Errno 110] Operation timed out
```

which looks like a failed syscall.

I'm using a fresh virtualenv with this pyocd and pyusb:

```
pyocd==0.23.0
pyusb==1.0.2
```

And I suspect libusb 1.0.23-2 from the arch packages is what pyusb is using as a backend, though I'm not sure.

@pabigot -- is there anything in `dmesg` or other likely suspects to narrow down the USB error?

mbolivar, 2019-12-11:
@pabigot are you by any chance running this in a virtual machine? Either way, you may want to look at 9c390d7655772dc5983624f71c22c17b9f9fa9c1, i.e. something like:

```
pyocd flash -da limit_packets=True -e sector -t nrf51 /mnt/nordic/zp/zephyr/samples/hello_world/build/zephyr/zephyr.hex
```

pabigot, 2019-12-11:
Not running in a VM, and that command doesn't help.

However, I did move the device from the USB 2.0 hub I've been using for years (that works fine for everything else) to a USB 3 hub, and it flashed just fine.  Once. Then I got a GPF from the cdc_acm driver which took out the entire bus.

In any case, it's clear this isn't a west problem or a pyocd problem, but probably Linux driver/hardware issue.  Thanks for the suggestions and checks.

closed 2019-12-11 by pabigot