zephyrproject-rtos/zephyr #21301: Coverage report generated for qemu_x86 board is incomplete

rmstoi, 2019-12-11:
I'm trying to generate coverage report for the hello_world sample.

To generate coverage reports I follow instructions at:

https://docs.zephyrproject.org/latest/guides/coverage.html

For native_posix it works just fine:

```
cmake -DBOARD=native_posix  -DCONFIG_COVERAGE=y  ..
lcov --capture --directory ./ --output-file lcov.info -q --rc lcov_branch_coverage=1
genhtml lcov.info --output-directory lcov_html -q --ignore-errors source --branch-coverage --highlight --legend

```

For qemu_x86 board the coverage report is incomplete, for example main.c is not included. It includes report only for files in "../../../lib/libc/minimal/include/". To generate qemu_x86 report:

```
cmake -DBOARD=qemu_x86  -DCONFIG_COVERAGE=y  ..
make run|tee log.log
python3 $ZEPHYR_BASE/scripts/gen_gcov_files.py -i log.log
mkdir gcov_report
gcovr -r $ZEPHYR_BASE . --html -o gcov_report/coverage.html --html-details --gcov-executable /opt/zephyr-sdk/x86_64-zephyr-elf/bin/x86_64-zephyr-elf-gcov
```

rmstoi, 2019-12-11:
OK, I checked the sanitycheck script, and the steps used there also work with qemu_x86 board

```
python3 $ZEPHYR_BASE/scripts/gen_gcov_files.py -i log.log
lcov --gcov-tool /opt/zephyr-sdk/i586-zephyr-elf/bin/i586-zephyr-elf-gcov --capture --directory . --rc lcov_branch_coverage=1 --output-file ./coverage.info
genhtml --legend --branch-coverage --ignore-errors source -output-directory coverage_html ./coverage.info
```

So, this would mean that our documentation is out-of-date

nashif, 2020-01-16:
The manual steps work for me just fine, I get more than libc

closed 2020-01-16 by nashif