zephyrproject-rtos/zephyr #21318: CONFIG_SYS_POWER_MANAGEMENT Makes Build Fail for nRF5340 and nRF9160

MarcianoPreciado, 2019-12-12:
The current implementation of `zephyr/subsys/power/device.c` breaks because of the following:

```
#if defined(CONFIG_SYS_POWER_MANAGEMENT)
...
/*
 * FIXME: Remove the conditional inclusion of
 * core_devices array once we enble the capability
 * to build the device list based on devices power
 * and clock domain dependencies.
 */
#if defined(CONFIG_SOC_SERIES_NRF52X) || defined(CONFIG_SOC_SERIES_NRF51X)
#define MAX_PM_DEVICES	15
#define NUM_CORE_DEVICES	4
#define MAX_DEV_NAME_LEN	16
static const char core_devices[NUM_CORE_DEVICES][MAX_DEV_NAME_LEN] = {
	"CLOCK_32K",
	"CLOCK_16M",
	"sys_clock",
	"UART_0",
};
#else
#error "Add SoC's core devices list for PM"
#endif
...
#endif /* defined(CONFIG_SYS_POWER_MANAGEMENT) */
```
The documentation describes that `SYS_POWER_MANAGEMENT` is dependent on `SOC_FAMILY_NRF` which is selected by `SOC_SERIES_NRF51X`, `SOC_SERIES_NRF52X`, `SOC_SERIES_NRF53X`, `SOC_SERIES_NRF91X`. Both of the soc_series symbols for the nRF53x and nRF91x select `HAS_SYS_POWER_STATE_DEEP_SLEEP_1` also which shows that the system power management functionality was intended to be compatible for both series.

I'm currently working on an extreme low-power iot device on the nRF5340 and this feature is necessary to reach the lowest power state without losing significant development time. I'd be happy to help via a pull request, but I am quite new to Zephyr and am not yet familiar enough to start modifying the codebase.

masz-nordic, 2019-12-12:
@MarcianoPreciado can you try #21075 and see if it helps?

MarcianoPreciado, 2019-12-13:
@masz-nordic I tried the pull request you mentioned and it seems to help a lot! Do you know exactly what is holding it up from being merged?

MarcianoPreciado, 2019-12-13:
I will close the issue once your pull request is accepted.

PR #21075 (2020-01-10): soc: nrf: basic power management for nrf53 and nrf91
Only System OFF mode is supported. Based on nrf52 power management, but nrf53 and nrf91 use nrf_regulators HAL instead of nrf_power.

Fixes #21318
closed 2020-01-10 by ioannisg