zephyrproject-rtos/zephyr #21321: error update for project civetweb

boonpin1975, 2019-12-12:
this error occurs when doing west update 

[image]

FWeissenb, 2019-12-16:
Same here.

`=== updating civetweb (modules\lib\civetweb):
fatal: ambiguous argument 'HEAD': unknown revision or path not in the working tree.
Use '--' to separate paths from revisions, like this:
'git <command> [<revision>...] -- [<file>...]'`

That's the error message not shown in the screenshot above.

jukkar, 2019-12-17:
Is this the same issues as #21426?

szundi, 2019-12-17:
> Is this the same issues as #21426?

workaround suggested not worked for me with this problem

boonpin1975, 2019-12-18:
this work around is not working on my system

carlescufi, 2019-12-18:
@szundi and @boonpin1975 can you please comment on https://github.com/zephyrproject-rtos/zephyr/issues/21426? I am closing this since it's clearly a duplicate.

closed 2019-12-18 by carlescufi

szundi, 2020-04-09:
Updated to newer git, deleted the folder that holds civetweb with all the others and this fixed it, after this the update worked. Before the delete it did not.