zephyrproject-rtos/zephyr #21322: Allow running a work queue on an existing thread

jmgao, 2019-12-12:
**Is your enhancement proposal related to a problem? Please describe.**
Right now, it's rather hard to repurpose the memory used by the main thread's stack after initialization. The obvious options are either to implement your own work queue, or write a pile of hacks to find the main stack, wait until the main thread has exited, and then spawn a work queue there. (The specific use case I have is to have a work queue that runs extremely-low priority work, so I can't use the system work queue, and I'm running out of memory and I have some stack-heavy initialization, which is why I want to run it on the main thread)

**Describe the solution you'd like**
Add a function that starts a work queue on the thread that it's called upon:
```extern void k_work_q_run(struct k_work_q *work_q)```

**Caveats**
The reason that this is a bug and not a 3 line pull request is that it doesn't seem obvious to me that there's a good way to synchronize between something that wants to submit to the main thread's work queue and the main thread actually getting far enough to initialize the work queue. This seems to be a more general problem, though: e.g. [this example](https://github.com/zephyrproject-rtos/zephyr/blob/master/samples/subsys/usb/hid/src/main.c) which initializes a delayed_work in main, but submits it in a callback which might be called before main finishes? Perhaps, instead of providing functionality to turn the main thread into a work queue, the main thread should always be a work queue, and the application's main being just a work item submitted to it?

aescolar, 2019-12-12:
CC @andrewboie @andyross

andyross, 2019-12-12:
Makes sense.  Or maybe `k_work_q_run_one_item()` and let the user control the main loop?  Seems like if you want to use your own thread for whatever reason, you probably have other things you want it to be doing.

Alternatively rework the scheme so the "work queue as ordered list of callback/data pairs" and "thread running a work queue" are different layers and you can mix and match as you want.

jmgao, 2019-12-13:
Adding a work queue to the main thread seemed like the best option after some thought. It seems unlikely that anyone is going to spawn their own thread that also want to have a work queue on: they can just create a work queue and submit a work item to it. This resolves the initialization order problems nicely, since we can be explicit about when the queue is ready.

pkral78, 2020-07-20:
+1
`bg_thread_main` devices initialization and setup calls need hundreds bytes from stack, we can reuse them for system queue.

dpkristensen, 2023-08-31:
I was interested in doing something like this, but I saw it was not implemented.  For what it's worth, I think @andyross 's suggestion of something like `k_work_q_run_one_item()` would work the best; it would be good if the work queue did not intrinsically rely on how the thread implements the wait.  Advantages to this would let me do things like have multiple priority levels of work queues to handle on the same thread, interleaving other non-work-queue items, etc...

One issue I see with the current implementation of `struct k_work` and `struct k_work_q` is that `struct k_work_q` uses a `struct k_thread` instead of `struct k_thread *`.  This means the concept of a work queue is intrinsically bound to the idea of a dedicated thread, and thus cannot easily be plugged into an existing thread.  Ideally nobody will be messing around with the internals of k_work or k_work_q, but I suspect a change will break some code somewhere.

To get this to run off of the main (or any other already existing thread), the main thread would have to create its thread instance inside of a `struct k_work_q`; the work queue would then exist regardless of whether anyone wanted it or not.

I think the only way to provide such functionality without the unwanted side-effects for all other threads is to create an alternate "swork" (think similar to "slist" and "snode") API which is **only** the queue implementation.

andyross, 2024-02-09:
This is straightforward and a useful improvement and should probably stay active.  I'm going to remove myself as an assignee for time reasons, but maybe we could consider this one of those good-first-project items that sometimes get collected.