zephyrproject-rtos/zephyr #21329: Submit module wasm-micro-runtime (WAMR) to zephyr

wenyongh, 2019-12-12:
## Introduction

WebAssembly is the cutting edge managed runtime technology that was created as a universal intermediate representation format, and initially for enabling programming web applications in multiple languages (C/C++/Rust/Go/Typescript..) and achieving near native performance. There is a strong industry trend to expand the usage of WebAssembly to IoT solutions.  

[WebAssembly Micro Runtime](https://github.com/bytecodealliance/wasm-micro-runtime/)  (WAMR) is a [Bytecode Alliance](https://bytecodealliance.org) open source project aiming to build a WebAssembly sandbox for embedded environment. The WAMR VMCore binary footprint is under 90K, and with the WAMR application library the total footprint is under 130K. The WAMR is potentially to bring  following dramatic changes to the traditional firmware application development experience: 

- Separate firmware application development
- Dynamical firmware application deployment
- Support more programming languages (C/C++/Rust/TypeScript..) 
- Sandbox environment to enable 3rd party applications

WAMR now is actively maintained by Intel together with the community.

### Problem description

We propose adopting WAMR as the compilation configurable module for Zephyr. The Zephyr developers can smoothly enable WAMR building in the Zephyr based system for the new experience of WASM application development.

### Proposed change

Use west tool to add WAMR under the modules/lib folder of the zephyrproject's source tree, add readme and a few samples. 

## Detailed RFC

(1) Fork WAMR repo (https://github.com/bytecodealliance/wasm-micro-runtime/) to zephyrproject-rtos (https://github.com/zephyrproject-rtos/).
(2) Add the related configuration in the manifest file west.yml to import WAMR.
(3) Add WAMR samples to zephyr repo (https://github.com/zephyrproject-rtos/zephyr).

### Proposed change (Detailed)

No change to source code of zephyr project. We need to ensure that WAMR source code can be cloned by the west tool and the WAMR samples can be compiled and run successfully.

### Dependencies

The change doesn't affect to zephyr project and other teams/projects.
The WAMR samples depend on lvgl project (https://github.com/zephyrproject-rtos/lvgl).

### Concerns and Unresolved Questions

None.

## Alternatives

None.

stephanosio, 2019-12-13:
I was thinking about doing something similar with the nanoFramework (.NET platform for microcontrollers), but I can see that this could potentially be a better solution.

carlescufi, 2019-12-13:
@wenyongh thank you for your proposal. In order for us to understand better what the implications of providing native WASM support to Zephyr via WAMR it would be perhaps interesting for you to present this during one of our TSC calls. @MaureenHelm is the chair person.

nashif, 2019-12-13:
Based on the posted PR there are no changes to Zephyr itself, the PR just add samples. It makes more sense to keep the samples as part of the WAMR project and just add a reference to them. This is a new class of modules that just integrates with Zephyr (without being part of Zephyr) we should be able to add on-demand, not sure if this should be forked or be part of the Zephyr project but we should be able to provide a link and a way to add it locally to the manifest in on-demand fashion. Thoughts?

xwang98, 2020-02-08:
Wenyong and I are the maintainers for WAMR project. Recently we have released the Ahead-of-Time compilation support for WAMR which makes the WASM application can run in nearly native speed. 

@MaureenHelm we will be happy to present it in the TSC calls. The only issue is the TSC meeting time is not friendly for our timezone (Beijing China). We wonder if we can have an additional one-time meeting for presenstation of WAMR?

wentongwu, 2020-03-25:
ping...

overheat, 2021-03-20:
Still not merged?

irgu, 2021-07-01:
Any updates on that topic? I would like to evaluate WAMR on Zephyr based nRF Connect SDK v1.5.1 (nRF52840-DK as hw platform) but it looks not straightforward currently.

carlescufi, 2021-07-16:
@wenyongh @xwang98 @wentongwu we would like to move this forward, but I cannot find the PR anymore. Would it be possible for either of you to follow the new procedure for integration of external code describe in the link below so we can take this a step forward?

https://docs.zephyrproject.org/latest/contribute/external.html

xwang98, 2021-07-16:
@carlescufi Happy to create a PR again after the long journey. :-)

carlescufi, 2021-07-19:
> @carlescufi Happy to create a PR again after the long journey. :-)

@xwang98 much appreciated, please do!

wenyongh, 2021-07-27:
@carlescufi Thanks a lot! I have created another PR #37229, could you please help review it? And to build the sample, we assume that WAMR is cloned under zephyrproject/modules/lib/wasm-micro-runtime, could you please help fork WAMR to https://github.com/zephyrproject-rtos so that I can modify west.yml for west to clone the code? Thanks.

wenyongh, 2021-10-19:
@irgu I have updated the PR according to the suggestions of reviewers to make it more generic, which makes WAMR as a module of Zephyr, so that developer can just write simple config file and few source codes, see [hello example prj.conf](https://github.com/zephyrproject-rtos/zephyr/pull/37229/files#diff-122f2c5bf0f55d636a583aa03b789e91dd6dca6a5f584aebb28dca90dafc9ba4) and [app-manager example prj.conf](https://github.com/zephyrproject-rtos/zephyr/pull/37229/files#diff-ebf863ec9bc2d26fbfb2b9f87f9c3eed1e25ee5dfb1462fa21e298fee13697a4). Now the PR is under review and there is no feedback yet, hope we can get some comments soon.

wenyongh, 2021-10-21:
Thanks @cfriedt, thanks @nashif, currently the samples suppose that the WAMR project is forked into https://github.com/zephyrproject-rtos and I have updated the zephyr/west.yml to add WAMR into its project list. If it is made as a git submodule of zephyr, I will refactor the samples.

cfriedt, 2021-10-21:
> Thanks @cfriedt, thanks @nashif, currently the samples suppose that the WAMR project is forked into https://github.com/zephyrproject-rtos and I have updated the zephyr/west.yml to add WAMR into its project list. If it is made as a git submodule of zephyr, I will refactor the samples.

It's not entirely necessary to merge the code directly into the Zephyr tree to be used by Zephyr, nor is it entirely necessary to fork WAMR.

See https://github.com/zephyrproject-rtos/example-application

nashif, 2024-02-12:
outdated, closing. Please open new issue with updated information if this is still needed.

closed 2024-02-12 by nashif