zephyrproject-rtos/zephyr #21335: net: TCP: Socket echo server does not accept incoming connections when TLS is enabled

doki-nordic, 2019-12-12:
**Describe the bug**
Socket echo server (`samples/net/sockets/echo_server`) does not accept incoming connections. This only happens when compiled with TLS support (e.g. with `overlay-tls.conf`). `git bisect` shown that first commit containing this problem was d88f25bd763e2dfa70873b3c2321f2f8677d643d by @jukkar created on Oct 17 (pull request #19919). After debugging I suspect that problem is somewhere inside `NET_CONN_CB(tcp_established)` in `tcp.c`. Simple workaround is to set `CONFIG_NET_TCP_AUTO_ACCEPT=y`.

**To Reproduce**
Steps to reproduce the behavior:
1. Build the sample
   ```
   west build -b qemu_x86 . -- -DCONF_FILE="prj.conf overlay-tls.conf"
   ```
2. Prepare [networking with QEMU](https://docs.zephyrproject.org/latest/guides/networking/qemu_setup.html)
3. Run the sample
   ```
   west build -t run
   ```
4. Connect using OpenSSL
   ```
   openssl s_client -connect 192.0.2.1:4242
   ```

**Expected behavior**
Valid connection with echo.

**Impact**
I suspect that it is generally not possible to connect to socket server with TLS.

**Screenshots or console output**
On OpenSSL side only:
```
CONNECTED(00000005)
```
and application waits indefinitely.

No additional output on qemu when connection starts (at least with default log level).

**Environment:**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA: d88f25bd763e2dfa70873b3c2321f2f8677d643d

**Additional context**
Also tested on HW: `nrf52840_pca10056` board - the same behavior.

jukkar, 2019-12-12:
Could you try #20628 if it fixes the issue in your setup?

doki-nordic, 2019-12-12:
> Could you try #20628 if it fixes the issue in your setup?

Yes, it fixes, but breaks case **without** TLS enabled.

Following description is about #20628 without TLS.

I see that connection is accepted - output from qemu:
```
[00:00:05.870,000] <inf> net_echo_server_sample: TCP (IPv4): Accepted connection
[00:00:05.870,000] <inf> net_echo_server_sample: Waiting for TCP connection on port 4242 (IPv4)...
```
but echo does not work. Echo server normally shows some logs, that it received some bytes, but with this PR there are no such logs, so the application does not receive any data.

Closing connection and connecting again shows following log:
```
[00:03:46.170,000] <err> net_echo_server_sample: Cannot accept more connections
[00:03:46.170,000] <inf> net_echo_server_sample: Waiting for TCP connection on port 4242 (IPv4)...
```

nvlsianpu, 2019-12-13:
cc @rlubos

jukkar, 2019-12-13:
@doki-nordic I think I figured out the culprit, can you try #21371 if it works for you

doki-nordic, 2019-12-13:
#21371 works

PR #21371 (2019-12-16): net: sockets: tls: Mark accepted socket correctly in accept()
The TCP code expects that we know when the socket has called accept()
in order to continue connection attempt.

Fixes #21335
commit b92a4c5c (2019-12-16): net: sockets: tls: Mark accepted socket correctly in accept()

The TCP code expects that we know when the socket has called accept()
in order to continue connection attempt.

Fixes #21335
closed 2019-12-16 by jukkar