zephyrproject-rtos/zephyr #21372: cc26x2r1_launchxl build passed, but can't flash

happymili, 2019-12-13:
### build
```
west build -p auto -b cc26x2r1_launchxl  samples/hello_world/
```
output
```
-- west build: build configuration:
       source directory: /home/linux/zephyrproject/zephyr/samples/hello_world
       build directory: /home/linux/zephyrproject/zephyr/build (created)
       BOARD: cc26x2r1_launchxl (origin: command line)
-- west build: generating a build system
Zephyr version: 2.1.0
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.7.3", minimum required is "3.4") 
-- Selected BOARD cc26x2r1_launchxl
-- Found west: /home/linux/.local/bin/west (found suitable version "0.6.3", minimum required is "0.6.0")
-- Loading /home/linux/zephyrproject/zephyr/boards/arm/cc26x2r1_launchxl/cc26x2r1_launchxl.dts as base
Devicetree configuration written to /home/linux/zephyrproject/zephyr/build/zephyr/include/generated/generated_dts_board.conf
Parsing Kconfig tree in /home/linux/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/linux/zephyrproject/zephyr/boards/arm/cc26x2r1_launchxl/cc26x2r1_launchxl_defconfig'
Merged configuration '/home/linux/zephyrproject/zephyr/samples/hello_world/prj.conf'
Configuration saved to '/home/linux/zephyrproject/zephyr/build/zephyr/.config'
-- The C compiler identification is GNU 9.2.0
-- The CXX compiler identification is GNU 9.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/linux/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Cache files will be written to: /home/linux/.cache/zephyr
-- Configuring done
-- Generating done
-- Build files have been written to: /home/linux/zephyrproject/zephyr/build
-- west build: building application
[3/115] Preparing syscall dependency handling

[110/115] Linking C executable zephyr/zephyr_prebuilt.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:        9288 B     360360 B      2.58%
      FLASH_CCFG:          88 B         88 B    100.00%
            SRAM:        3816 B        80 KB      4.66%
        IDT_LIST:          40 B         2 KB      1.95%
[115/115] Linking C executable zephyr/zephyr.elf
```

### flash
```
west flash
```
output
```
-- west flash: rebuilding
ninja: no work to do.
-- west flash: using runner openocd
-- runners.openocd: Flashing file: /home/linux/zephyrproject/zephyr/build/zephyr/zephyr.hex
Open On-Chip Debugger 0.10.0+dev-01231-g3d7db5535-dirty (2019-11-20-13:42)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter_nsrst_delay: 100

Info : XDS110: connected
Info : XDS110: firmware version = 3.0.0.5
Info : XDS110: hardware version = 0x0023
Info : XDS110: connected to target via JTAG
Info : XDS110: TCK set to 2500 kHz
Info : clock speed 2500 kHz
Error: JTAG scan chain interrogation failed: all zeroes
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: cc26x2.jrc: IR capture error; saw 0x02 not 0x01
Warn : Bypassing JTAG setup events due to errors
Info : Listening on port 3333 for gdb connections
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* cc26x2.cpu         cortex_m   little cc26x2.cpu         tap-disabled

Info : TAP cc26x2.jrc does not have IDCODE
Info : JTAG tap: auto0.tap tap/device found: 0x3bb4102f (mfg: 0x017 (Texas Instruments), part: 0xbb41, ver: 0x3)
Error: cc26x2.jrc: IR capture error; saw 0x02 not 0x01
Warn : Bypassing JTAG setup events due to errors
Error: Target not examined yet
Error: auto_probe failed
auto erase enabled
Info : XDS110: disconnected
ERROR: command exited with status 1: /home/linux/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/openocd -s /home/linux/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts -f /home/linux/zephyrproject/zephyr/boards/arm/cc26x2r1_launchxl/support/openocd.cfg -c init -c targets -c 'reset halt' -c 'flash write_image erase /home/linux/zephyrproject/zephyr/build/zephyr/zephyr.hex' -c 'reset halt' -c 'verify_image /home/linux/zephyrproject/zephyr/build/zephyr/zephyr.hex' -c 'reset run' -c shutdown
```

vanti, 2019-12-18:
@happymili Could you double-check you have jumpers on TDO and TDI on the board, which are necessary for JTAG connections?

happymili, 2019-12-26:
@vanti I can use TI UniFlash to download hex file, and I can also use IAR & CCS to debug the development board. So, I think the hardware connection is OK.
Thanks for your reply.

vanti, 2020-01-08:
@happymili Sorry for the delay, just got back from vacation. Which version of OpenOCD are you using? Are you using the one from Zephyr SDK 0.10.3 (https://docs.zephyrproject.org/latest/getting_started/installation_linux.html)?

I tried to reproduce your issue using this setup:

- commit d385d86f7412f16ea8ed7027a51cd45b4f3cf9ef
- Zephyr SDK 0.10.3
- CC2652R1 LaunchPad with a sticker on the back that says "HW Rev B, FW: SDK2.40, 1847"

Using the command you used to build, I was successful in loading to my cc2652R1 board with "west flash" (the error messages are a known issue - the code runs fine afterwards):

```
$ west flash
-- west flash: rebuilding
ninja: no work to do.
-- west flash: using runner openocd
-- runners.openocd: Flashing file: /home/local/zephyrproject-work/zephyr/build/zephyr/zephyr.hex
Open On-Chip Debugger 0.10.0+dev-00992-g3333261df-dirty (2019-08-16-00:14)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 2500 kHz
srst_only separate srst_gates_jtag srst_open_drain connect_deassert_srst
adapter_nsrst_delay: 100
Info : XDS110: connected
Info : XDS110: firmware version = 2.3.0.16
Info : XDS110: hardware version = 0x0023
Info : XDS110: connected to target via JTAG
Info : XDS110: TCK set to 2500 kHz
Info : clock speed 2500 kHz
Info : JTAG tap: cc26x2.jrc tap/device found: 0x3bb4102f (mfg: 0x017 (Texas Instruments), part: 0xbb41, ver: 0x3)
Info : JTAG tap: cc26x2.cpu enabled
Info : cc26x2.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : Listening on port 3333 for gdb connections
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* cc26x2.cpu         cortex_m   little cc26x2.cpu         unknown
Info : JTAG tap: cc26x2.jrc tap/device found: 0x3bb4102f (mfg: 0x017 (Texas Instruments), part: 0xbb41, ver: 0x3)
Info : JTAG tap: cc26x2.cpu enabled
Error: Debug regions are unpowered, an unexpected reset might have happened
Error: JTAG-DP STICKY ERROR
Error: Could not find MEM-AP to control the core
target halted due to debug-request, current mode: Thread 
xPSR: 0x61000000 pc: 0x1000118e msp: 0x11001ff0
auto erase enabled
wrote 360448 bytes from file /home/local/zephyrproject-work/zephyr/build/zephyr/zephyr.hex in 6.666206s (52.804 KiB/s)
Info : JTAG tap: cc26x2.jrc tap/device found: 0x3bb4102f (mfg: 0x017 (Texas Instruments), part: 0xbb41, ver: 0x3)
Info : JTAG tap: cc26x2.cpu enabled
Error: Debug regions are unpowered, an unexpected reset might have happened
Error: JTAG-DP STICKY ERROR
Error: Could not find MEM-AP to control the core
target halted due to debug-request, current mode: Thread 
xPSR: 0x61000000 pc: 0x1000118e msp: 0x11001ff0
verified 360448 bytes in 4.754256s (74.039 KiB/s)
Info : JTAG tap: cc26x2.jrc tap/device found: 0x3bb4102f (mfg: 0x017 (Texas Instruments), part: 0xbb41, ver: 0x3)
Info : JTAG tap: cc26x2.cpu enabled
Error: Debug regions are unpowered, an unexpected reset might have happened
Error: JTAG-DP STICKY ERROR
Error: Could not find MEM-AP to control the core
shutdown command invoked
Info : XDS110: disconnected
```

seanmlyons22, 2020-01-14:
@happymili , I am seeing the same issue as you.
Seeing this with both the [Zephyr fork](https://github.com/zephyrproject-rtos/openocd) of openocd as well as the [TI OPENOCD Package](http://www.ti.com/tool/SIMPLELINK-OPENOCD).

On zephyr revision `d24e5e520c`.

We have very similar versions of the Emulator pack firmware on the XDS110.

```
➜  zephyr git:(master) west flash
-- west flash: rebuilding
ninja: no work to do.
-- west flash: using runner openocd
-- runners.openocd: Flashing file: /Users/seanlyons/dev/zephyrproject/zephyr/build/zephyr/zephyr.hex
Open On-Chip Debugger 0.10.0+dev-00936-g0a13ca1a8 (2020-01-14-21:43)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter_nsrst_delay: 100

Info : XDS110: connected
Info : XDS110: firmware version = 3.0.0.7
Info : XDS110: hardware version = 0x0023
Info : XDS110: connected to target via JTAG
Info : XDS110: TCK set to 2500 kHz
Info : clock speed 2500 kHz
Error: JTAG scan chain interrogation failed: all zeroes
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: cc26x2.jrc: IR capture error; saw 0x00 not 0x01
Warn : Bypassing JTAG setup events due to errors
Info : Listening on port 3333 for gdb connections
    TargetName         Type       Endian TapName            State
--  ------------------ ---------- ------ ------------------ ------------
 0* cc26x2.cpu         cortex_m   little cc26x2.cpu         tap-disabled

Info : TAP cc26x2.jrc does not have IDCODE
Info : JTAG tap: auto0.tap tap/device found: 0x3bb4102f (mfg: 0x017 (Texas Instruments), part: 0xbb41, ver: 0x3)
Error: cc26x2.jrc: IR capture error; saw 0x02 not 0x01
Warn : Bypassing JTAG setup events due to errors
Error: Target not examined yet
Error: auto_probe failed
auto erase enabled
Info : XDS110: disconnected
```

seanmlyons22, 2020-01-14:
[This post](https://e2e.ti.com/support/wireless-connectivity/sub-1-ghz/f/156/t/837129?tisearch=e2e-sitesearch&keymatch=launchxl-cc1312r1) seems to confirm the suspicion that it is related to the emupack f/w differences from TI.

Until openocd is updated to support this new f/w, is there a way to modify the CLI args that west uses when invoking openocd?

vanti, 2020-01-15:
@seanmlyons22 I think the post you are referring to is on the right track. @happymili, as a workaround, could you try to add the following line at the bottom of _boards/arm/cc26x2r1_launchxl/support/openocd.cfg_:

`adapter_khz 1500`

I upgraded to firmware 3.0.0.5 and the above allows me to flash successfully with 'west flash'.

PR #23303 (2020-03-10): Openocd workaround for cc13x2/cc26x2
Newer XDS firmware fails to flash correctly with higher adapter clock
speed. Using a slower speed for now to prevent errors.

Fixes #21372
commit 13b10b54 (2020-03-10): boards: cc26x2r1_launchxl: workaround for xds firmware 3.x

Newer XDS firmware fails to flash correctly with higher adapter clock
speed. Using a slower speed for now to prevent errors.

Fixes #21372
commit 2d5ed2a9 (2020-03-10): boards: cc1352r1_launchxl: workaround for xds firmware 3.x

Newer XDS firmware fails to flash correctly with higher adapter clock
speed. Using a slower speed for now to prevent errors.

Fixes #21372
closed 2020-03-10 by jhedberg