zephyrproject-rtos/zephyr #21378: The program cannot be downloaded to nrf52840, only to pca10056

ZhenfeiWu, 2019-12-13:
The program cannot be downloaded to nrf52840, only to pca10056

I made a program about nrf52840 small system. When I turn off and restart the small system, I can't start zephyr

carlescufi, 2019-12-13:
@ZhenfeiWu can you please explain a bit more exactly what you are trying to do?

ZhenfeiWu, 2019-12-14:
thank you for your replay，
my question is：
I have made a nrf52840 small system board, such as the hex generated by compiling in / basic / Binky. I can download the program through DK, and my board can work normally alone. But I can download the program through DK only by keeping my board connected with DK.
why?

ZhenfeiWu, 2019-12-14:
thank you for your replay,
I have made a nrf52840 small system board, such as the hex generated by compiling in / basic / Binky. My board can work normally independently by downloading the program through DK, but like the hex generated by compiling in / Bluetooth / mesh/Demo, if I download the program through DK, I must keep my board connected with DK to work normally, and cannot work normally independently.
why?
thank you very much

carlescufi, 2020-07-10:
This is stale, and there's not enough information. Please reopen if still applicable.

closed 2020-07-10 by carlescufi