zephyrproject-rtos/zephyr #21379: Bluetooth: Mesh: Node Reset Not Clear Bind Key Information

LingoMeng, 2019-12-13:
After receive node reset message, local node exec `bt_mesh_reset`
mod->flag will be fill in some values, but `bt_mesh_comp_unprovision` clear those mod->flag cause https://github.com/zephyrproject-rtos/zephyr/blob/b3da9efbec586bdac0c61671c189b1b6f13eb4ae/subsys/bluetooth/mesh/settings.c#L1556-L1577 not executed.
https://github.com/zephyrproject-rtos/zephyr/blob/43f833800ce9da195985133d070455bef8ddc5b0/subsys/bluetooth/mesh/main.c#L111-L154

aescolar, 2019-12-13:
CC @jhedberg

jhedberg, 2019-12-13:
@LingoMeng so you're saying the model configuration doesn't end up being removed from flash although the rest of the network configuration is? Are you planning to submit a PR for this? Btw, I didn't see you quote the exact place where mod->flags gets cleared. Your second quote contains bt_dev.flags which is a different variable. @trond-snekvik FYI.

LingoMeng, 2019-12-13:
Sorry, maybe I initialized this value in the model initialization callback. `model_init[i].init`,

closed 2019-12-13 by LingoMeng