zephyrproject-rtos/zephyr #21390: BLE Incomplete Connect results in subsquent encryption failures

lanvis, 2019-12-13:
**Describe the bug**
After a failed incomplete connect sequence, I get the following warning.
 "<wrn> bt_conn: Found valid but disconnected conn object"

The connect is successful but when I attempt to encrypt the "bonded" link 
bt_conn_set_security(conn, BT_SECURITY_L2)
it is successful but  I get the following warning:
**<wrn> bt_att: No pending ATT request**

When I check the key size: bt_conn_enc_key_size(conn) it returns 0 so encryption fails.

Zephyr 2.0, Central BLE.

**To Reproduce**
Put device in an RF quiet box while attempting to connect to a peripheral.

**Expected behavior**
It should connect and encrypt without error and encryption key should be > 0

**Impact**
deveopment phase

**Screenshots or console output**
Ellisys trace attached.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK

[FalcFoundValidButDisconnectedObjece_Plus_PendingATTRequest.zip](https://github.com/zephyrproject-rtos/zephyr/files/3962756/FalcFoundValidButDisconnectedObjece_Plus_PendingATTRequest.zip)

jhedberg, 2019-12-14:
> bt_conn: Found valid but disconnected conn object

This sounds like a left-over  bt_conn reference from a previous connection (possibly in the application). Which application are you using? Are you able to reproduce it with any of the samples in the Zephyr tree?

lanvis, 2019-12-30:
Custom app.
To secure link, I use the conn from the successful connect.

joerchan, 2020-01-03:
@lanvis Please check for possible bt_conn_ref missing unrefs.
Probably the unref from the encryption procedure is not released when disconnected.

closed 2020-03-18 by nashif