zephyrproject-rtos/zephyr #21410: bt_ctlr_hci: Tx Buffer Overflow on r LL/CON/MAS/BV-04-C, LL/CON/SLA/BV-05-C & LL/CON/SLA/BV-06-C

aescolar, 2019-12-16:
**Describe the bug**
At some point during the last couple of week in master LL/CON/MAS/BV-04-C, LL/CON/SLA/BV-05-C & LL/CON/SLA/BV-06-C started failing 

**To Reproduce**
The fastest way, would be to
1. take this branch: https://github.com/zephyrproject-rtos/zephyr/pull/21006
2. west update
3. source zephyr-env.sh
4. WORK_DIR=/persistant/scratch/alpi/Zephyr/west_zephyr/bsim_bt_out tests/bluetooth/bsim_bt/compile.sh
5. tests/bluetooth/bsim_bt/edtt_ble_test_app/tests_scripts/ll.1.sh
6. See error in the log

To narrow the run to just the failing case, you can edit
tests/bluetooth/bsim_bt/edtt_ble_test_app/tests_scripts/ll.set1.test_list
to only include one of those failing tests (just delete or comment all other lines)

**Expected behavior**
No errors, testcase passing

**Impact**
It would seem the controller will not pass certification

**Screenshots or console output**
```
LL/CON/MAS/BV-04-C [Master receiving Data packets from Slave] test started...
d_01: @00:03:52.498407  [00:03:52.498,382] <inf> bt_ctlr_hci: handle: 0x0000, rssi: -35 dB.
d_02: @00:03:52.498404  [00:03:52.498,382] <inf> bt_ctlr_hci: handle: 0x0000, rssi: -35 dB.
d_02: @00:03:52.855773  Failed to send ACL Data (err -22)
d_02: @00:03:53.135192  Failed to send ACL Data (err -22)
d_02: @00:03:53.414611  Failed to send ACL Data (err -22)
d_02: @00:03:53.694030  Failed to send ACL Data (err -22)
d_02: @00:03:53.973449  Failed to send ACL Data (err -22)
d_02: @00:03:54.252868  Failed to send ACL Data (err -22)
d_02: @00:03:54.512329  Failed to send ACL Data (err -22)
d_02: @00:03:54.791748  [00:03:54.791,748] <err> bt_ctlr_hci: Tx Buffer Overflow
d_02: @00:03:54.791748  Failed to send ACL Data (err -105)
d_02: @00:03:54.851623  [00:03:54.851,623] <err> bt_ctlr_hci: Tx Buffer Overflow
d_02: @00:03:54.851623  Failed to send ACL Data (err -105)
d_02: @00:03:54.911499  [00:03:54.911,499] <err> bt_ctlr_hci: Tx Buffer Overflow
d_02: @00:03:54.911499  Failed to send ACL Data (err -105)
d_02: @00:03:54.971374  [00:03:54.971,374] <err> bt_ctlr_hci: Tx Buffer Overflow
d_02: @00:03:54.971374  Failed to send ACL Data (err -105)
```

**Environment:**
 - OS: Linux (Ubuntu 18.04)
 - Toolchain: host gcc
 - Commit SHA: (today's master) 9a111df676851838cf635ed79d38d23fed4b46a9

cvinayak, 2019-12-18:
The said test cases LL/CON/MAS/BV-04-C, LL/CON/SLA/BV-05-C & LL/CON/SLA/BV-06-C, all passed on EBQ tester 2019-1.7263

aescolar, 2019-12-18:
Fixed by https://github.com/EDTTool/EDTT/pull/1

closed 2019-12-18 by aescolar