zephyrproject-rtos/zephyr #21426: civetweb triggers an error on Windows with Git 2.24

carlescufi, 2019-12-16:
**Describe the bug**
When running `west update` using Git 2.24 on Windows, you get:

```
fatal: filename in tree entry contains backslash: '\'
error: https://github.com/zephyrproject-rtos/civetweb did not send all necessary objects
```

**To Reproduce**

On Windows, and with Git 2.24:
```
west init zephyr
cd zephyr
west update
```

**Environment (please complete the following information):**
 - OS: Windows
 - Git for Windows 2.24
 - Commit: civetweb 99129c5efc907ea613c4b73ccff07581feb58a7a

mbolivar, 2019-12-19:
Users on the duplicate issue (#21321) reported the `git config --global core.protectNTFS` workaround did not work for them.

I would recommend:

1. apply the `git config work around`
2. backing up any local commits, then deleting the civetweb directory (modules/lib/civetweb) entirely
3. retrying `west update`

andrewboie, 2020-01-02:
A note about this in our documentation would be nice, I tripped over this the other day.

mbolivar, 2020-01-02:
> A note about this in our documentation would be nice, I tripped over this the other day.

The patch fixing it appears to be making its way through review on mainline git and git-for-windows, so hopefully a new version with the fix will be out soon and we can just close this.

carlescufi, 2020-01-07:
Closing this issue. See workarounds in the description

closed 2020-01-07 by carlescufi

szundi, 2020-03-28:
just a tip: if you get the error with git 2.26+, delete modules/lib/civetweb folder, then retry with west update, this fixed me some bad state