zephyrproject-rtos/zephyr #21441: Add UART5 on B-port to H7 pinmux

parmentj, 2019-12-17:
**Is your enhancement proposal related to a problem? Please describe.**
In the provided zephyr/drivers/pinmux/stm32/pinmux_stm32h7.h, the use of UART5 on PB12 and PB13 is missing (Alternate function 14).

Add on line 69 of pinmux_stm32h7.h :
 #define STM32H7_PINMUX_FUNC_PB13_UART5_TX   \
 	(STM32_PINMUX_ALT_FUNC_14 | STM32_PUSHPULL_PULLUP)

 #define STM32H7_PINMUX_FUNC_PB12_UART5_RX    \
 	(STM32_PINMUX_ALT_FUNC_14 | STM32_PUSHPULL_NOPULL)

**Describe alternatives you've considered**
I added the above into my pinmux.c

**Additional context**
-

carlescufi, 2019-12-17:
@parmentj why don't you send a Pull Request with the change?

parmentj, 2019-12-18:
@carlescufi , stage fright , I am going through the description right now 

I will make it so later today

closed 2020-03-30 by MaureenHelm