zephyrproject-rtos/zephyr #21443: "HCI_USB" sample doesn't compile with "nucleo_wb55rg" board

Hekkfern, 2019-12-17:
**Description** 
Using "west" to compile the example "hci_usb" (located in samples/bluetooth) fails with the "nucleo_wb55rg". However, it compiles with the "nrf52840_pca10056" board for example.

**To Reproduce** 
Steps to reproduce the behavior:
1. Open terminal
2. Go to the "zephyr" project path.
3. Execute zephyr_env.sh to create ZEPHYR_TOOLCHAIN_VARIANT, ZEPHYR_BASE and ZEPHYR_SDK_INSTALL_DIR variables.
4. west build -p auto -b nucleo_wb55rg samples/bluetooth/hci_usb
5. Compilation fails

**Environment:** 
 - OS: Ubuntu 18.04 LTS
 - Zephyr library version 2.1.99 

**Console output** 
-- west build: making build dir /home/hfernandez/Projects/zephyrproject/zephyr/build pristine
-- west build: build configuration:
       source directory: /home/hfernandez/Projects/zephyrproject/zephyr/samples/bluetooth/hci_usb
       build directory: /home/hfernandez/Projects/zephyrproject/zephyr/build
       BOARD: nucleo_wb55rg (origin: command line)
-- west build: generating a build system
Zephyr version: 2.1.99
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.9", minimum required is "3.4") 
-- Selected BOARD nucleo_wb55rg
-- Found west: /home/hfernandez/.local/bin/west (found suitable version "0.6.3", minimum required is "0.6.0")
-- Loading /home/hfernandez/Projects/zephyrproject/zephyr/boards/arm/nucleo_wb55rg/nucleo_wb55rg.dts as base
Devicetree configuration written to /home/hfernandez/Projects/zephyrproject/zephyr/build/zephyr/include/generated/generated_dts_board.conf
Parsing Kconfig tree in /home/hfernandez/Projects/zephyrproject/zephyr/samples/bluetooth/hci_usb/Kconfig
Loaded configuration '/home/hfernandez/Projects/zephyrproject/zephyr/boards/arm/nucleo_wb55rg/nucleo_wb55rg_defconfig'
Merged configuration '/home/hfernandez/Projects/zephyrproject/zephyr/samples/bluetooth/hci_usb/prj.conf'
Configuration saved to '/home/hfernandez/Projects/zephyrproject/zephyr/build/zephyr/.config'
-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /opt/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Cache files will be written to: /home/hfernandez/.cache/zephyr
CMake Warning at ../../../subsys/usb/CMakeLists.txt:27 (message):
  CONFIG_USB_DEVICE_VID has default value 0x2FE3.

  This value is only for testing and MUST be configured for USB products.

-- Configuring done
-- Generating done
-- Build files have been written to: /home/hfernandez/Projects/zephyrproject/zephyr/build
-- west build: building application
[1/148] Preparing syscall dependency handling

[63/148] Building C object zephyr/CMakeFiles/z...phyr.dir/drivers/usb/device/usb_dc_stm32.c.obj
FAILED: zephyr/CMakeFiles/zephyr.dir/drivers/usb/device/usb_dc_stm32.c.obj 
ccache /opt/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DBUILD_VERSION=zephyr-v2.1.0-210-g5f629c892993 -DCORE_CM4 -DHSE_VALUE=32000000 -DKERNEL -DSTM32WB55xx -DUSE_FULL_LL_DRIVER -DUSE_HAL_DRIVER -D_FORTIFY_SOURCE=2 -D__PROGRAM_START -D__ZEPHYR__=1 -I../include -I../include/drivers -Izephyr/include/generated -I../soc/arm/st_stm32/stm32wb -I../drivers -I../ext/hal/cmsis/Core/Include -I../subsys/usb -I../subsys/bluetooth -I/home/hfernandez/Projects/zephyrproject/modules/hal/stm32/stm32cube/stm32wbxx/soc -I/home/hfernandez/Projects/zephyrproject/modules/hal/stm32/stm32cube/stm32wbxx/drivers/include -I/home/hfernandez/Projects/zephyrproject/modules/hal/stm32/stm32cube/stm32wbxx/drivers/include/Legacy -I/home/hfernandez/Projects/zephyrproject/modules/hal/stm32/lib/stm32wb/hci -isystem ../lib/libc/minimal/include -isystem /opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/include -isystem /opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/include-fixed -Os -imacros/home/hfernandez/Projects/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -mthumb -mcpu=cortex-m4 -imacros/home/hfernandez/Projects/zephyrproject/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/home/hfernandez/Projects/zephyrproject/zephyr/samples/bluetooth/hci_usb=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/hfernandez/Projects/zephyrproject/zephyr=ZEPHYR_BASE -ffunction-sections -fdata-sections -mabi=aapcs -march=armv7e-m -std=c99 -nostdinc -MD -MT zephyr/CMakeFiles/zephyr.dir/drivers/usb/device/usb_dc_stm32.c.obj -MF zephyr/CMakeFiles/zephyr.dir/drivers/usb/device/usb_dc_stm32.c.obj.d -o zephyr/CMakeFiles/zephyr.dir/drivers/usb/device/usb_dc_stm32.c.obj   -c /home/hfernandez/Projects/zephyrproject/zephyr/drivers/usb/device/usb_dc_stm32.c
/home/hfernandez/Projects/zephyrproject/zephyr/drivers/usb/device/usb_dc_stm32.c:153:44: error: 'DT_USB_NUM_BIDIR_ENDPOINTS' undeclared here (not in a function)
  struct usb_dc_stm32_ep_state out_ep_state[DT_USB_NUM_BIDIR_ENDPOINTS];
                                            ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/hfernandez/Projects/zephyrproject/zephyr/drivers/usb/device/usb_dc_stm32.c: In function 'usb_dc_stm32_clock_enable':
/home/hfernandez/Projects/zephyrproject/zephyr/drivers/usb/device/usb_dc_stm32.c:199:10: error: 'DT_USB_CLOCK_BUS' undeclared (first use in this function); did you mean 'DT_ADC_1_CLOCK_BUS'?
   .bus = DT_USB_CLOCK_BUS,
          ^~~~~~~~~~~~~~~~
          DT_ADC_1_CLOCK_BUS
/home/hfernandez/Projects/zephyrproject/zephyr/drivers/usb/device/usb_dc_stm32.c:199:10: note: each undeclared identifier is reported only once for each function it appears in
/home/hfernandez/Projects/zephyrproject/zephyr/drivers/usb/device/usb_dc_stm32.c:200:10: error: 'DT_USB_CLOCK_BITS' undeclared (first use in this function); did you mean 'DT_SPI_2_CLOCK_BITS'?
   .enr = DT_USB_CLOCK_BITS,
          ^~~~~~~~~~~~~~~~~
          DT_SPI_2_CLOCK_BITS
In file included from ../include/arch/arm/irq.h:18,
                 from ../include/arch/arm/arch.h:27,
                 from ../include/arch/cpu.h:17,
                 from ../include/kernel_includes.h:34,
                 from ../include/kernel.h:17,
                 from ../include/device.h:11,
[... 37 lines cut ...]
                                  LL_APB1_GRP1_PERIPH_LCD
/home/hfernandez/Projects/zephyrproject/zephyr/drivers/usb/device/usb_dc_stm32.c: In function 'usb_dc_ep_configure':
/home/hfernandez/Projects/zephyrproject/zephyr/drivers/usb/device/usb_dc_stm32.c:556:6: error: 'DT_USB_RAM_SIZE' undeclared (first use in this function); did you mean 'DT_SRAM_SIZE'?
  if (DT_USB_RAM_SIZE <=
      ^~~~~~~~~~~~~~~
      DT_SRAM_SIZE
In file included from ../include/arch/arm/irq.h:17,
                 from ../include/arch/arm/arch.h:27,
                 from ../include/arch/cpu.h:17,
                 from ../include/kernel_includes.h:34,
                 from ../include/kernel.h:17,
                 from ../include/device.h:11,
                 from ../include/drivers/usb/usb_dc.h:20,
                 from ../include/usb/usb_device.h:39,
                 from /home/hfernandez/Projects/zephyrproject/zephyr/drivers/usb/device/usb_dc_stm32.c:49:
/home/hfernandez/Projects/zephyrproject/zephyr/drivers/usb/device/usb_dc_stm32.c: In function 'usb_dc_ep_write':
/home/hfernandez/Projects/zephyrproject/zephyr/drivers/usb/device/usb_dc_stm32.c:722:15: error: 'DT_USB_IRQ' undeclared (first use in this function); did you mean 'TSC_IRQn'?
   irq_disable(DT_USB_IRQ);
               ^~~~~~~~~~
../include/irq.h:267:43: note: in definition of macro 'irq_disable'
 #define irq_disable(irq) arch_irq_disable(irq)
                                           ^~~
[68/148] Building C object zephyr/arch/arch/ar...iles/arch__arm__core__cortex_m.dir/fault.c.obj
ninja: build stopped: subcommand failed.
ERROR: command exited with status 1: /usr/bin/cmake --build /home/hfernandez/Projects/zephyrproject/zephyr/build

jfischer-phytec-iot, 2019-12-17:
USB is not supported on nucleo_wb55rg

closed 2019-12-17 by jfischer-phytec-iot

erwango, 2019-12-18:
@Hekkfern, indeed usb is not yet supported on nucleo_wb55rg.
Though, USB IP on STM32WB is similar to other STM32 USB IP supported on zephyr.
So if you're interested, you can find below an example on how USB could enabled on STM32WB:
https://github.com/zephyrproject-rtos/zephyr/pull/18806/commits/83585fe034fd307b43b23c91be2a8ab8effa89e1
Then, you can check USB related changes in following commit to enable support on nucleo_wb55rg:
https://github.com/zephyrproject-rtos/zephyr/pull/18806/commits/590621e644691644b09d6fd77407b7864cf126fd