zephyrproject-rtos/zephyr #21445: drivers/i2c: add I2C slave support for nrfx

sdalu, 2019-12-17:
On the nordic side, at least nrf52832 and nrf52840 have I2C slave peripheral, zephyr seems to also have an internal API for I2C slave support (in drivers/i2c.h), but driver implementation is currently missing

anangl, 2019-12-17:
It is not feasible to use nRF TWIS peripherals via the I2C slave API that is currently available in Zephyr. This API requires the I2C transactions to be performed in a byte-by-byte manner while TWIS periherals use EasyDMA transfers for whole transactions. That's why there is no driver implementing this API.
However, the nrfx driver for TWIS peripherals ([nrfx_twis](https://github.com/zephyrproject-rtos/hal_nordic/blob/master/nrfx/drivers/include/nrfx_twis.h)) can be used directly in Zephyr. In order to do it one needs to:
- enable the driver for the desired TWIS instance by selecting the corresponding `NRFX_TWIS*` [option](https://github.com/zephyrproject-rtos/zephyr/blob/master/modules/Kconfig.nordic#L314)
  _for TWIS0 this will be (in `prf.conf`):_
  ```
  CONFIG_NRFX_TWIS0=y
  ```
- enable in devicetree the corresponding `i2c*` node to be used as TWIS
  _for TWIS0 it will be (in `<board>.overlay`, pin numbers are just sample ones):_
  ```
  &i2c0 {
  	compatible = "nordic,nrf-twis";
  	status = "okay";
  	sda-pin = <26>;
  	scl-pin = <27>;
  };
  ```
- include `<nrfx_twis.h>` in the source file that needs to call this driver's API
- install the proper IRQ handler for the selected TWIS instance by calling `IRQ_CONNECT`;
  _for TWIS0 it will be:_
  ```
  IRQ_CONNECT(DT_NORDIC_NRF_TWIS_I2C_0_IRQ_0,
  	    DT_NORDIC_NRF_TWIS_I2C_0_IRQ_0_PRIORITY,
  	    nrfx_isr, nrfx_twis_0_irq_handler, 0);
  ```

sdalu, 2020-01-22:
A bit of topic, do you have examples of nrfx used for twis ?

hoai265, 2020-01-22:
> A bit of topic, do you have examples of nrfx used for twis ?

Just do as @anangl comment bro.
My **prj.conf**

```
CONFIG_NRFX_TWIS=y
CONFIG_NRFX_TWIS0=y
CONFIG_NRFX_PRS=y
```

I added an overlay file **nrf52811_pca10056.overlay**.

```
&i2c0 {
	compatible = "nordic,nrf-twis";
	status = "okay";
	sda-pin = <3>;
	scl-pin = <4>;
};
```

And after that i use the old nRF SDK I2C slave code with calling IRQ_CONNECT in initialize function.
This is my initialize example function:      
 ```
 IRQ_CONNECT(DT_NORDIC_NRF_TWIS_I2C_0_IRQ_0, 
                             DT_NORDIC_NRF_TWIS_I2C_0_IRQ_0_PRIORITY, 
                             nrfx_isr, nrfx_twis_0_irq_handler, 0);
 nrfx_twis_config_t config;
 config.addr[0] = SLAVE_ADDR;
 config.addr[1] = 0;
 config.scl = PIN_SCL;
 config.scl_pull = NRF_GPIO_PIN_PULLUP;
 config.sda = PIN_SDA;
 config.sda_pull = NRF_GPIO_PIN_PULLUP;
 config.interrupt_priority = 2;
 ret = nrfx_twis_init(&m_twis, &config, twis_event_handler);
 nrfx_twis_enable(&m_twis);
```

SJFOM, 2020-06-30:
Hi there,

I'm trying to implement the above on an nrf52832 dev board (PCA10040). I'm getting "not found" errors for the `DT_NORDIC_NRF_TWIS_I2C_0_IRQ_0` and  `DT_NORDIC_NRF_TWIS_I2C_0_IRQ_0_PRIORITY` flags. My main.c file is below: 

```
#include <zephyr.h>
#include <device.h>
#include <devicetree.h>
#include <drivers/gpio.h>
#include <nrfx_twi.h>
#include <nrfx_twis.h>
#include <irq.h>

/* 1000 msec = 1 sec */
#define SLEEP_TIME_MS   1000

/* The devicetree node identifier for the "led0" alias. */
#define LED0_NODE DT_ALIAS(led0)

#if DT_NODE_HAS_STATUS(LED0_NODE, okay)
#define LED0	DT_GPIO_LABEL(LED0_NODE, gpios)
#define PIN	DT_GPIO_PIN(LED0_NODE, gpios)
#if DT_PHA_HAS_CELL(LED0_NODE, gpios, flags)
#define FLAGS	DT_GPIO_FLAGS(LED0_NODE, gpios)
#endif
#else
/* A build error here means your board isn't set up to blink an LED. */
#error "Unsupported board: led0 devicetree alias is not defined"
#define LED0	""
#define PIN	0
#endif

#ifndef FLAGS
#define FLAGS	0
#endif

#define SLAVE_ADDR (0x70 >> 1U)

typedef nrfx_twis_t         nrf_drv_twis_t;

static const nrfx_twis_t m_twis = NRFX_TWIS_INSTANCE(0);

static void twis_event_handler(nrfx_twis_evt_t const * const p_event)
    {
        switch (p_event->type)
        {
        case NRFX_TWIS_EVT_READ_REQ:
            if (p_event->data.buf_req)
            {
                // do something
            }
            break;
        case NRFX_TWIS_EVT_READ_DONE:
            break;
        case NRFX_TWIS_EVT_WRITE_REQ:
            if (p_event->data.buf_req)
            {
                // do something
            }
            break;
        case NRFX_TWIS_EVT_WRITE_DONE:
            break;

        case NRFX_TWIS_EVT_READ_ERROR:
        case NRFX_TWIS_EVT_WRITE_ERROR:
        case NRFX_TWIS_EVT_GENERAL_ERROR:
            break;
        default:
            break;
        }
    }

void main(void)
{
	struct device *dev;
	bool led_is_on = true;
	int ret;

	dev = device_get_binding(LED0);
	if (dev == NULL) {
		return;
	}

	ret = gpio_pin_configure(dev, PIN, GPIO_OUTPUT_ACTIVE | FLAGS);
	if (ret < 0) {
		return;
	}

        IRQ_CONNECT(DT_NORDIC_NRF_TWIS_I2C_0_IRQ_0, 
                            DT_NORDIC_NRF_TWIS_I2C_0_IRQ_0_PRIORITY, 
                            nrfx_isr, nrfx_twis_0_irq_handler, 0);

        nrfx_twis_config_t config;
        config.addr[0] = SLAVE_ADDR;
        config.addr[1] = 0;
        config.scl = 27; //PIN_SCL;
        config.scl_pull = NRF_GPIO_PIN_PULLUP;
        config.sda = 26; //PIN_SDA;
        config.sda_pull = NRF_GPIO_PIN_PULLUP;
        config.interrupt_priority = 2;
        ret = nrfx_twis_init(&m_twis, &config, twis_event_handler);
        nrfx_twis_enable(&m_twis);

	while (1) {
		gpio_pin_set(dev, PIN, (int)led_is_on);
		led_is_on = !led_is_on;
                printk("ret value: %d\n", ret);
		k_msleep(SLEEP_TIME_MS);
	}
}
```

My overlay file is:

```
&i2c0 {
	compatible = "nordic,nrf-twis";
	status = "okay";
	sda-pin = <26>;
	scl-pin = <27>;
};
```

My prj.conf file is:

```
CONFIG_GPIO=y
CONFIG_NRFX_TWIS=y
CONFIG_NRFX_TWIS0=y
# CONFIG_NRFX_PRS=y
```

I cannot include `CONFIG_NRFX_PRS` as Segger complains that `soltuion load command failed (1)` when I do.

Any pointers here would be immensely appreciated!

Sam

anangl, 2020-06-30:
> I'm trying to implement the above on an nrf52832 dev board (PCA10040). I'm getting "not found" errors for the `DT_NORDIC_NRF_TWIS_I2C_0_IRQ_0` and `DT_NORDIC_NRF_TWIS_I2C_0_IRQ_0_PRIORITY` flags.

@SJFOM Use `DT_IRQN(DT_NODELABEL(i2c0))` and `DT_IRQ(DT_NODELABEL(i2c0), priority)` instead.

SJFOM, 2020-06-30:
Hi @anangl, that worked a treat - many thanks!

pabigot, 2020-08-19:
Per above comment the existing API cannot be supported for NRF, so I'm going to close this.  #27675 tracks the larger issue; support on Nordic hardware will be part of that effort.

closed 2020-08-19 by pabigot