zephyrproject-rtos/zephyr #21448: nrf52840 errata_98 / 89 mixup

aep, 2019-12-17:
the code for errata_98 and errata_89 is identical, this doesn't seem right.

errata 98 is "[98] NFCT: Not able to communicate with the peer" and the workaround is different in the code and in the docs.

https://infocenter.nordicsemi.com/index.jsp?topic=%2Ferrata_nRF52840_EngA%2FERR%2FnRF52840%2FEngineeringA%2Flatest%2Ferr_840.html&cp=4_0_1_5

```
Write 0x00038148 to 0x4000568C before the NFC peripheral is enabled:

 *(volatile uint32_t *)0x4000568Cul = 0x00038148ul; 
```

i found this because i'm getting a usage fault when accessing 0x10000130ul in errata_89. likely totally unrelated, but mentioning it just in case it is.

anangl, 2019-12-17:
> the code for errata_98 and errata_89 is identical, this doesn't seem right.

The code in `errata_96` and `errata_97` is also identical to the one in `errata_89`. But there is nothing wrong in it. These functions check if a given workaround needs to be applied. And the workaround for nRF52840 anomaly 98 is applied in [system_nrf52840.c](https://github.com/zephyrproject-rtos/hal_nordic/blob/master/nrfx/mdk/system_nrf52840.c#L104).

aep, 2019-12-17:
ah! thanks for the explanation. not sure how that works, but glad its correct

closed 2019-12-17 by aep