zephyrproject-rtos/zephyr #21450: `sample.net.cloud.google_iot_mqtt` test is failing for `frdm_k64f`

carlocaione, 2019-12-17:
The `samples/net/cloud/google_iot_mqtt/sample.net.cloud.google_iot_mqtt` test fails to build for the `frdm_k64f` platform. Currently reproducible on master.

```
$ ./scripts/sanitycheck -p frdm_k64f --testcase-root samples/net/cloud/google_iot_mqtt/
...
Zephyr version: 2.1.99
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.7.5", minimum required is "3.4") 
-- Selected BOARD frdm_k64f
-- Found west: /home/carlo/.local/bin/west (found suitable version "0.6.2", minimum required is "0.6.0")
-- Loading /home/carlo/work/proj/zephyr/zephyrproject/zephyr/boards/arm/frdm_k64f/frdm_k64f.dts as base
Devicetree configuration written to /home/carlo/work/proj/zephyr/zephyrproject/zephyr/sanity-out/frdm_k64f/samples/net/cloud/google_iot_mqtt/sample.net.cloud.google_iot_mqtt/zephyr/include/generated/generated_dts_board.conf
Parsing Kconfig tree in /home/carlo/work/proj/zephyr/zephyrproject/zephyr/samples/net/cloud/google_iot_mqtt/Kconfig
Loaded configuration '/home/carlo/work/proj/zephyr/zephyrproject/zephyr/boards/arm/frdm_k64f/frdm_k64f_defconfig'
Merged configuration '/home/carlo/work/proj/zephyr/zephyrproject/zephyr/samples/net/cloud/google_iot_mqtt/prj.conf'
Configuration saved to '/home/carlo/work/proj/zephyr/zephyrproject/zephyr/sanity-out/frdm_k64f/samples/net/cloud/google_iot_mqtt/sample.net.cloud.google_iot_mqtt/zephyr/.config'
-- The C compiler identification is GNU 9.2.0
-- The CXX compiler identification is GNU 9.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/carlo/work/proj/zephyr/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Cache files will be written to: /home/carlo/.cache/zephyr
CMake Error at CMakeLists.txt:9 (message):
  !!!!!! Generate key file before continuing.  See README !!!!!

-- Configuring incomplete, errors occurred!
See also "/home/carlo/work/proj/zephyr/zephyrproject/zephyr/sanity-out/frdm_k64f/samples/net/cloud/google_iot_mqtt/sample.net.cloud.google_iot_mqtt/CMakeFiles/CMakeOutput.log".
See also "/home/carlo/work/proj/zephyr/zephyrproject/zephyr/sanity-out/frdm_k64f/samples/net/cloud/google_iot_mqtt/sample.net.cloud.google_iot_mqtt/CMakeFiles/CMakeError.log".
```

This has been caught by shippable on #21326.

carlocaione, 2019-12-17:
This is probably not a bug per se because the key can be generated using the provided `create_keys.py` script but not sure how that is supposed to work on the CI side.

jukkar, 2019-12-17:
I just sent a PR that disables the compilation of this sample by sanitychecker. We should make it possible to compile the sample automatically using some pre-defined certs so that the sample can be compile tested. That is for future studies.

PR #21451 (2019-12-17): samples: net: google_iot_mqtt: Do not compile by sanitycheck
The sample needs key.c file but that cannot be generated by
sanitychecker. So disable compilation by sanitycheck.
Eventually we should make it possible to compile the sample
using some pre-defined values so that the sample will not
bit-rot but that is for later.

Fixes #21450
commit 51fc680b (2019-12-17): samples: net: google_iot_mqtt: Do not compile by sanitycheck

The sample needs key.c file but that cannot be generated by
sanitychecker. So disable compilation by sanitycheck.
Eventually we should make it possible to compile the sample
using some pre-defined values so that the sample will not
bit-rot but that is for later.

Fixes #21450
closed 2019-12-17 by jukkar