zephyrproject-rtos/zephyr #21452: drivers: ethernet: unify the initiaization

alexanderwachter, 2019-12-17:
**Is your enhancement proposal related to a problem? Please describe.**
ETH_NET_DEVICE_INIT takes a device-init function, and an API with an iface_api init.
The device-init, which is called first, should initialize the hardware to a state where it is ready to send and receive.

The net interface-init function should only bind the device driver to the interface and enable receiving.

At the moment, not all of the drivers follow this procedure.

**Describe the solution you'd like**
All drivers should initialize the same way.

**Additional context**
Drivers that follow this procedure (MAC address is set in the interface-init function):
- e1000
- liteeth
- native posix
- smsc911x

Drivers that follow this procedure (MAC address is set in the device-init function):
- stellaris (MAC address is set in device_init)
- mcux (MAC address is set in device_init)

Drivers that are fully initialized with receiving enabled after device-init (MAC address is set in the device-init function):
- enc28j60
- enc424j600

Drivers that do hardware initialization in the interface-init function:
- sam_gmac
- stm32_hal

pabigot, 2020-07-07:
API 2020-07-07: Triage review shows now evidence that this has been discussed.  Leaving in Triage with intent it should be on the agenda for 2020-07-14.

pabigot, 2020-07-07:
@alexanderwachter if this issue is no longer of interest to you, please close this instead.

alexanderwachter, 2020-07-14:
@pabigot I don‘t have any particular interest in this issue. It was a finding during a discussion. I‘m not sure if this issue is causing any problems, but it does not feel right if the drivers behave differently.

carlescufi, 2020-07-14:
@jukka or @tbursztyka do you have interest in moving this forward, or should we close it?

jukkar, 2020-07-20:
I think we can close this, the current drivers work and both me and @tbursztyka are currently busy with other stuff.

closed 2020-07-20 by jukkar