zephyrproject-rtos/zephyr #21455: driver: subsys: sdhc: USAGE FAULT trace and no cs control

ric96, 2019-12-17:
**Describe the bug**
sdhc driver crashes post initialization and mount.

Log is from the fat_fs sample.

The line `... 2OK3OK4OK3OK4OK3 ...` are just some printk I added around the k_sleep while loop. So it looks like the loop executes for some time. I also changed the k_sleep value to 100ms.

The cs control also seems to be inhibited, not sure why as the correct pin b12 is mentioned in the overlay file.

Board: 96Boards Carbon
Tested ~5 different sdcards all with same result and two different sd-to-spi modules.

LOG:
```
*** Booting Zephyr OS build zephyr-v2.1.0-349-g51fc680ba225  ***
[00:00:00.005,000] <inf> spi_ll_stm32: CS control inhibited (no GPIO device)
Sector size 512
Memory Size(MB) 1910
Mounting Disk 
Disk mounted.

Listing dir /SD: ...
[DIR ] A
[DIR ] S
[DIR ] F
[DIR ] G
[FILE] B (size = 0)
[FILE] C (size = 0)
[FILE] D (size = 0)
OKOK2OK3OK4OK3OK4OK3OK4OK3OK4OK3OK4OK3OK4OK3OK4OK3OK4OK3OK4OK3[00:00:00.030,000] <inf> sdhc_spi: Found a ~1910 MiB SDHC card.
[00:00:00.031,000] <inf> sdhc_spi: Manufacturer ID=27 OEM='SM' Name='00000' Revision=0x10 Serial=0x28f2183b
[00:00:00.031,000] <inf> main: Block count 3911680
[00:00:00.059,000] <inf> sdhc_spi: Found a ~1910 MiB SDHC card.
[00:00:00.060,000] <inf> sdhc_spi: Manufacturer ID=27 OEM='SM' Name='00000' Revision=0x10 Serial=0x28f2183b
[00:00:01.030,000] <err> os: ***** USAGE FAULT *****
[00:00:01.030,000] <err> os:   Illegal load of EXC_RETURN into PC
[00:00:01.030,000] <err> os: r0/a1:  0x00000000  r1/a2:  0xe000ed00  r2/a3:  0x200005c4
[00:00:01.030,000] <err> os: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x0800695b
[00:00:01.030,000] <err> os:  xpsr:  0x00000000
[00:00:01.030,000] <err> os: Faulting instruction address (r15/pc): 0x00000000
[00:00:01.030,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:01.030,000] <err> os: Current thread: 0x200002a8 (unknown)
[00:00:01.125,000] <err> os: Halting system
```

**To Reproduce**
Steps to reproduce the behavior:
1. west build samples/subsys/fs/fat_fs/ -b 96b_carbon 
2. west flash
4. See error

**Expected behavior**
No output after `<inf> sdhc_spi: Manufacturer ID=27 OEM='SM' Name='00000' Revision=0x10 Serial=0x28f2183b`.

**Environment (please complete the following information):**
 - OS: Linux, Fedora 31
 - Toolchain Zephyr SDK 10.3
 - Commit: 51fc680ba225c2df5f3fca9064839b825a9c0511

erwango, 2019-12-18:
@ric96 since I'm not able to reproduce, would you be able to provide more information, like a backtrace for instance?
Would you have any way to reproduce on non STM32 HW to see if this is related to SPI driver ?

ric96, 2019-12-18:
I don't have any suitable non stm32 micros at the moment.

So here is a very verbosely logged output using the following configs: https://pastebin.ubuntu.com/p/wStzbYM6pg/

```
CONFIG_LOG=y
CONFIG_LOG_OVERRIDE_LEVEL=4
CONFIG_EXCEPTION_STACK_TRACE=y
CONFIG_LOG_BUFFER_SIZE=65536
CONFIG_DEBUG=y
CONFIG_DEBUG_INFO=y
CONFIG_OMIT_FRAME_POINTER=n

```

jhedberg, 2020-02-18:
@ric96 do you still see this issue with the latest tree?

closed 2020-03-18 by nashif

EdytaBosacka, 2020-07-29:
@ric96 Did you fix this?