zephyrproject-rtos/zephyr #21466: doc: extract_content.py not copying images in a table

dbkinder, 2019-12-18:
**Describe the bug**
The doc build script ``extract_content.py`` is used to scan .rst files looking for referenced files (images and includes for example) and copies them to the proper location in the _build folder.

The script is missing such references if they're within a drawn table (as encountered in PR #21361, and doesn't copy the referenced images, resulting in a doc build error that the images can't be found.

I don't think the script recognized the ``.. image`` directive because there's a pipe symbol earlier on the line with the directive, and there are also two image directives on the same line (not properly handled in the script either, I suspect).

```
    +-------------------------------------+-------------------------------------------+
    | .. image:: ./spidongle_assy6791.png | .. image:: ./spidongle_assy6791_view2.png |
    |      :width: 300px                  |      :width: 380px                        |
    |      :align: center                 |      :align: center                       |
    |      :alt: SPI DONGLE ASSY 6791     |      :alt: SPI DONGLE ASSY 6791 view 2    |
    |                                     | .. image:: ./dediprog_connector_2.png     |
    |                                     |      :width: 380px                        |
    |                                     |      :align: center                       |
    |                                     |      :alt: SPI DONGLE ASSY 6791 Connected |
    |                                     |                                           |
    +-------------------------------------+-------------------------------------------+
```

Workaround was to use a list-table directive (which did work), but the script should be updated to handle this valid use of images in a table.

dbkinder, 2019-12-18:
@carlescufi @mbolivar Not a common problem, but it was interesting to diagnose.

carlescufi, 2019-12-18:
@dbkinder thanks for the report. I've assigned it to @ulfalizer who has been doing a lot of work lately with the documentation Python scripts.

ulfalizer, 2019-12-19:
Fix: https://github.com/zephyrproject-rtos/zephyr/pull/21509

PR #21509 (2019-12-20): doc: extract_content.py: Detect included files in a more robust way
Allow `.. <figure/include/image/...>:: <path>` to appear anywhere within
a line, and find multiple directives on a single line. This is needed to
find files included e.g. within tables.

Implemented by making the `<path>` part of the regex more specific and
searching for matches anywhere within the contents of the file. Should
be a bit faster too.

Maybe there's some tiny potential for false positives, but this
generates the same file list as the old version for the current docs at
least.

Fixes: #21466
commit ea3041db (2019-12-20): doc: extract_content.py: Detect included files in a more robust way

Allow '.. <figure/include/image/...>:: <path>' to appear anywhere within
a line, and find multiple directives on a single line. This is needed to
find files included e.g. within tables.

Implemented by making the <path> part of the regex more specific and
searching for matches anywhere within the contents of the file. Should
be a bit faster too.

Maybe there's some tiny potential for false positives, but this
generates the same file list as the old version for the current docs at
least.

Fixes: #21466
closed 2019-12-20 by nashif