zephyrproject-rtos/zephyr #21469: ARC SMP is untested in sanitycheck

andrewboie, 2019-12-18:
ARC has support for SMP now, but there are no ARC-based boards which have CONFIG_MP_NUM_CPUS or CONFIG_SMP set in their defconfigs. I would expect nsim_hs_smp to have CONFIG_SMP=y in it.

Thus, all tests are running uniprocessor instead of in SMP mode except for those two tests that enable SMP specifically. We are not adequately testing the ARC SMP implementation.

This needs to be fixed, if we have an emulated ARC target that supports SMP this must be enabled by default in its defconfig, much like how qemu_x86_64 does.

andrewboie, 2019-12-18:
@vonhust can you please look into this?

abrodkin, 2019-12-18:
I think @vonhust point was `CONFIG_SMP` get automatically enabled by SMP-related tests & examples.
But I guess @andrewboie point here is we need to run all the tests with SMP enabled just to see how good SMP implementation is, right?

Now the problem is free version of nSIM (which requires no license whatsoever) doesn't support SMP so the only way to try SMP on ARC is to run tests on real HSDK board. And we may share 1 or 2 with the community but then we need to figure out who's going to use them. Another option would be to automatically run sanity checking against master (and/or whichever identified as useful other branches) in Synopsys and somehow report results.

andrewboie, 2019-12-18:
There's only one test which forces SMP to be on: tests/kernel/smp. It does not by itself comprehensively catch all the things that can go wrong in SMP, for example see #21317 which we are currently fighting 

> Now the problem is free version of nSIM (which requires no license whatsoever) doesn't support SMP

Is the forthcoming QEMU implementation of ARC going to support SMP?
How much does an SMP-enabled nSIM license cost?

> the only way to try SMP on ARC is to run tests on real HSDK board. And we may share 1 or 2 with the community but then we need to figure out who's going to use them.

We could certainly use some over here, our validation team could integrate them into the HW testing we are doing.

>  Another option would be to automatically run sanity checking against master (and/or whichever identified as useful other branches) in Synopsys and somehow report results.

This works for me too.

PR #22362 (2020-03-12): bug  fixes in arc smp initializaiton and tests
This PR is designed to fix  issue #21469. It fixes bugs related to SMP:
   * initialization
   * timer driver
   * test cases
   * board configuration

Because sanitycheck test does not support mdb which is required by ARC SMP, I mannually run some critical tests and samples, the resultes are ok.

Fixes #21469
closed 2020-03-12 by nashif