zephyrproject-rtos/zephyr #21489: Allow to read any types during discovery

rakons, 2019-12-18:
**Is your enhancement proposal related to a problem? Please describe.**
We have some module to handle service discovery. This module does the whole discovery, caches all the outputs and when it finishes, it provides easy interface for the application. For a reference you can see it here (https://github.com/NordicPlayground/fw-nrfconnect-nrf/blob/master/subsys/bluetooth/gatt_dm.c).

Now we wish to add CEP support directly into it.
To get all the CEP attributes values I wish to use **Read By Type Request**. I could use ``bt_gatt_read`` with ``params::handle_count`` set to 0, and I should get what I wish, but it would force me to add ``bt_gatt_read_params`` structure data into my instance. But all the data I really need is still there in my ```bt_gatt_discovery_params`` instance. To limit memory usage and data copying I wish to have a possibility to read CEP using my ``bt_gatt_discovery_params``.

**Describe the solution you'd like**
Implement the required functionality into discovery function.
We have all the data which we need: uuid is here, start and end handle is also here. We just need additional functionality like BT_GATT_DISCOVER_READ.

**Describe alternatives you've considered**
We can align discovery and read data, it seems quite similar, but only include data inside discovery seems to break the interoperability:

** 2: Align read and discover structures layout **
Then we do not need anything more, and probably we do not need to modify the code.
Just add a possibility to create discovery and read structures union with all the critical data in place.

Lets look how it looks now:
```
struct bt_gatt_read_params {
	struct bt_att_req _req;
	bt_gatt_read_func_t func;
	size_t handle_count;
	union {
		struct {
			u16_t handle;
			u16_t offset;
		} single;
		u16_t *handles;
		struct {
			u16_t start_handle;
			u16_t end_handle;
			struct bt_uuid *uuid;
		} by_uuid;
	};
};
```

```
struct bt_gatt_discover_params {
	struct bt_att_req _req;
	/** Discover UUID type */
	struct bt_uuid *uuid;
	/** Discover attribute callback */
	bt_gatt_discover_func_t func;
	union {
		struct {
			/** Include service attribute declaration handle */
			u16_t attr_handle;
			/** Included service start handle */
			u16_t start_handle;
			/** Included service end handle */
			u16_t end_handle;
		} _included;
		/** Discover start handle */
		u16_t start_handle;
	};
	/** Discover end handle */
	u16_t end_handle;
	/** Discover type */
	u8_t type;
};
```
If we can move somehow the _included at the end, rest of the fields seems to be easily organized to match the others: uuid needs to be moved, and type from discovery would be placed together with handle_count from read.

But still - the _included inside discovery structure messes it a lot - not sure if it may be easy done.

**Additional context**
NC

Vudentz, 2019-12-19:
I have plans to make a pool of bt_att_req so we don't have to embedded them into the params, that would make them quite more slim, the other thing we have been planning is to support general discovery procedure, in that case you would just provide the service UUID you are interested in and it would do the discovery of all attributes within the service range when a general discovery is triggered.

I probably wouldn't go as far as to create a cache for the entire database since to make use of it in a generic form it would take a lot of memory to store everything, so it might just provice the attributes with handle + UUID and fetch things like CEP if available, but perhaps it is possible to have a cache just for the UUIDs registered but that doesn't eliminate the size problem some services can have multiple instances.

rakons, 2019-12-19:
@Vudentz - Look how it is done in the discovery manager I have mentioned above. We cache only the parameters of one service detected. Then you are allowed to continue and process next service.

I am not sure now if should I continue with CEP implementation if there is going to be new API - did you already schedule some timing when you wish to implement it?

joerchan, 2019-12-19:
@Vudentz Are you saying then that we should have a way to read CEP using `bt_gatt_discovery_params` or not?

The size optimization of the params is not really a blocker for this feature request. Do you suggest to do it in that order anyway, and do you intend to both yourself?

Leaving caching of the discovered attributes and values up to the application is perfectly fine. Don't need to have that in the host, the use-case decides what needs to be stored.

Vudentz, 2019-12-30:
> @Vudentz Are you saying then that we should have a way to read CEP using `bt_gatt_discovery_params` or not?

That would be a different discovery type that would incorporate reading CEP.

> The size optimization of the params is not really a blocker for this feature request. Do you suggest to do it in that order anyway, and do you intend to both yourself?

I guess it makes sense to do in that order so we don't end up having to reworking the discovery once we introduce a general discovery type.

> Leaving caching of the discovered attributes and values up to the application is perfectly fine. Don't need to have that in the host, the use-case decides what needs to be stored.

It would be perfectly fine if only the application would use the cache but the stack might actually be interested in services as well, for instaces things like GATT, GAP, and other services that the stack itself may have client for. Also for things like CEP it makes sense that the stack has helpers to read it since otherwise applications may have to duplicating code for doing it on their own.

PR #32585 (2021-04-06): Bluetooth: host: Add discovery of standard descriptor values
Add discovery of standard descriptor values. This provides a shorthand
instead of doing a read by UUID, allowing the application discovery
procedure to re-use the discover parameters for simple descriptors.

Fixes: #21489
commit 9f7061a8 (2021-04-06): Bluetooth: host: Add discovery of standard descriptor values

Add discovery of standard descriptor values. This provides a shorthand
instead of doing a read by UUID, allowing the application discovery
procedure to re-use the discover parameters for simple descriptors.

Fixes: #21489
closed 2021-04-06 by nashif