zephyrproject-rtos/zephyr #21498: Zephyr Peripheral not responding to Terminate command from central

kcnsunrise, 2019-12-18:
I'm testing a zephyr ble HRM peripheral. The test consists of sending a Feature Exchange Command from a central and then terminating the connection from the central before the peripheral can respond to the command. According to the spec, the peripheral should terminate the connection when the terminate command is received. What I see is the peripheral ignores the terminate command and sends the Feature Request Response. 

[Feature Exchange Termination Latest Zephyr Peripheral.zip](https://github.com/zephyrproject-rtos/zephyr/files/3980438/Feature.Exchange.Termination.Latest.Zephyr.Peripheral.zip)

Here's a message exchange from slack.

Joakim Andersson  9 days ago
@vich @Kim Nisbet It looks like a failure on the central side. The central sends LLCP Disconnect. Peripheral NACKs, central sends PHY REQ.

Joakim Andersson  9 days ago
No wait. It is the disconnect packet that is nacking the feature response. So slave ACKs the disconnect. But continues to listen, and central continues with PHY_REQ.

Joakim Andersson  9 days ago
Event 257
FEATURE_REQ -->
            <-- EMPTY
Event 258
EMPTY       -->
            <-- FEATURE_RSP
DISCONNECT  --> (nack)
            <-- FEATURE_RSP (retransmit)
** Link should have been disconnected by both at this point **
Event 259
PHY_REQ     -->
            <-- EMPTY

Joakim Andersson  9 days ago
Failure in both devices it seems.

aescolar, 2019-12-19:
CC @cvinayak @joerchan @jhedberg

cvinayak, 2019-12-24:
Please share the commit hash, samples used and your modified controller implementation that is forcing a second feature exchange (or something that is causing the controller to send feature request when it has already done it once in the current connection).

carlescufi, 2020-02-14:
@kcnsunrise could you please add information as to how to reproduce this issue as requested by @cvinayak ?

carlescufi, 2020-07-09:
Closing due to inactivity, Please reopen if it's still an issue.

closed 2020-07-09 by carlescufi