zephyrproject-rtos/zephyr #21511: re-visit k_thread_abort wrt SMP

andrewboie, 2019-12-19:
The default k_thread_abort implementation has this note:

```
void z_impl_k_thread_abort(k_tid_t thread)
{
	/* We aren't trying to synchronize data access here (these
	 * APIs are internally synchronized).  The original lock seems
	 * to have been in place to prevent the thread from waking up
	 * due to a delivered interrupt.  Leave a dummy spinlock in
	 * place to do that.  This API should be revisted though, it
	 * doesn't look SMP-safe as it stands.
	 */
	struct k_spinlock lock = {};
	k_spinlock_key_t key = k_spin_lock(&lock);

	__ASSERT((thread->base.user_options & K_ESSENTIAL) == 0U,
		 "essential thread aborted");

	z_thread_single_abort(thread);
	z_thread_monitor_exit(thread);

	if (thread == _current && !arch_is_in_isr()) {
		z_swap(&lock, key);
	} else {
		/* Really, there's no good reason for this to be a
		 * scheduling point if we aren't aborting _current (by
		 * definition, no higher priority thread is runnable,
		 * because we're running!).  But it always has been
		 * and is thus part of our API, and we have tests that
		 * rely on k_thread_abort() scheduling out of
		 * cooperative threads.
		 */
		z_reschedule(&lock, key);
	}
}
```

Opening this ticket for further contemplation and resolution.

andrewboie, 2019-12-19:
> /* We aren't trying to synchronize data access here (these
>  * APIs are internally synchronized).  The original lock seems
>  * to have been in place to prevent the thread from waking up
>  * due to a [...]

Have a couple questions about this:

- This spinlock on the stack effectively just locks interrupts on the current CPU. But there's no synchronization if some other CPU is trying to abort this thread, or indeed perform any operation on the thread object
- My initial guess was that this ought to acquire the spinlock in thread.c, I think functions like z_thread_single_abort() assume that lock is held. But if I grab that lock instead I get many test timeouts so that's not it.
- wondering if we should re-visit the overall kernel locking and make it data-centric; have a core kernel lock, and per-thread locks, maybe separate locks for run queues, etc, seems like the locks we have now are closely associated with the C file they are in but we really want to protect data and not code

andyross, 2019-12-19:
Yeah, I was looking at this yesterday.  In fact the call to ready a thread out of k_thread_create() and to abort a thread here are both basically under the wrong lock.  They really ought to move under the scheduler lock for their core work.  Though I couldn't see an obvious spot where there's a clear race, either.

andrewboie, 2019-12-19:
> Though I couldn't see an obvious spot where there's a clear race, either.

Yeah, the only thing that looks clearly dangerous so far is two threads on different CPUs calling k_thread_abort() on the same object (or a thread naturally exiting while another CPU tries to call k_thread_abort() on it)

But nothing pops out to suggest it's the ghost we're chasing in #21317

nashif, 2020-08-14:
@andrewboie  that is another k_thread_abort issue...

andrewboie, 2020-08-14:
Gonna close this. There's no lock here any more, and the underlying thread exit stuff is all serialized on the sched spinlock. The issues with self-aborts racing with external aborts is known and can be resolved as part of #26486. Closing for now.

closed 2020-08-14 by andrewboie