zephyrproject-rtos/zephyr #21512: Add k_pipe_flush API

koffes, 2019-12-19:
It would be beneficial to be able to flush a k_pipe. Suggest adding a k_pipe_flush() API.

AFAIK k_pipe_cleanup() will free a dyn allocated buffer, but this will in many cases be too harsh. The k_pipe_flush() call will keep the buffer allocated, but clear the contents.

aescolar, 2019-12-19:
CC @andrewboie @andyross

peter-mitsis, 2021-10-01:
I've started taking a look at the pipe kernel objects. Here is how I am currently interpreting the enhancement request as it does not say anything about writers that are blocked.

A pipe flush would empty the contents of the pipe buffer (if applicable) and unblock all the blocked writers on that pipe. All data (both in the pipe and waiting to go into the pipe) is lost, but those previously pended writers think it succeeded. Blocked readers (if any) are unaffected. Allocated buffers are not freed.

koffes, 2021-10-07:
As it is quite some time since I created this issue, I do not recall all details. Why would the pending data be lost?

peter-mitsis, 2021-10-07:
Glad you asked. Admittedly, it might not be the best or ideal proposal (losing pending data), but it is at least a starting point to establish what we want the behaviour to be.

The model I had in my head was that of a clogged kitchen sink--the pipe is clogged/full (this is the pipe buffer) and there is water waiting to drain (this is all the blocked writers).  In this model, flushing the pipe not only removes the clog, but it allows all the water in the sink to drain too--otherwise the pipe (buffer) fills again and there may still be water in the sink (waiting writers). The lost data would be because in the flush, all the write data is not going to a reader; it would be like writing everything to /dev/null--it just goes into empty non-fillable space.

The above model is does differ from  k_pipe_cleanup() which was mentioned enhancement. k_pipe_cleanup() appears to only perform the cleanup if there are neither readers nor writers pending on the pipe--so it impacts only the allocated buffer (which also has the side effect of transforming the buffered pipe into a bufferless pipe--albeit broken).

The ultimate question is "What behaviour do we want/need from a pipe flush?"

PR #39597 (2022-01-10): Adds a k_pipe_flush() API
Fixes #21512

This set of commits adds two pipe flush APIs: k_pipe_buffer_flush() and k_pipe_flush().

k_pipe_buffer_flush() clears out just the specified pipe's ring buffer. If there were threads pending to write into this buffer, then they will attempt write into the buffer just as if another thread had issued a k_pipe_get() call and read those same bytes from the pipe. Thus, it is analogous to reading buffer size bytes into a temporary and then discarding the buffer (but without using a temporary buffer).

k_pipe_flush() builds upon this idea and flushes not just a [... cut]
closed 2022-01-10 by nashif