zephyrproject-rtos/zephyr #21513: NULL parameter checks in Zephyr APIs

koffes, 2019-12-19:
Zephyr contains numerous APIs which require a pointer to be supplied as parameter which is not NULL.
Usually, the doc specifies this, but as we all know, developers make mistakes. 
When a NULL pointer is supplied where it should not, may lead to an MPU_FAULT.  When an MPU_FAULT occurs, this often requires more comprehensive knowledge of embedded systems, such as *.elf files, the addr2line tool and so forth to find the root cause.
For many of us, ease of development and customer satisfaction is key. For a user to get direct LOG output if NULL is supplied using e.g. __LINE__ and __FILE__ would greatly ease development.

In order not to bloat production code, I suggest making a NULL_PARAM_CHECK(var1, var2, ...) macro which in the inner workings prints __LINE__, __FILE__, flushes the LOG and ASSERTS.  Alternatively, instead of asserting, invoking k_panic(). This param check macro can be enabled by default in DEBUG mode.

This issue has already been partially discussed here #19473.

ceolin, 2020-03-20:
@andrewboie @koffes what exactly ? I mean, don't we have this features with __ASSERT ?
This macro even allows application decides which action takes.

koffes, 2020-03-20:
@ceolin We can absolutely use __ASSERT. Creating a separate macro was just an option. This issue was created in a broader sense, in that many places are missing checks for NULL pointers.

carlescufi, 2020-03-31:
Anything that supports userspace (syscalls) already verifies parameters when actually calling from userspace into kernel mode. Outside of userspace this is not done consistently, and could be done using the existing `CHECKIF()`.

closed 2020-03-31 by carlescufi